robotology/mesh-support #1430: [R1SN003] - Left wrist broken

ste93, 2022-09-20:
### Device name 🤖

R1SN003

### Request/Failure description

Unfortunately the left wrist is broken, it seems that one of the part of the tripod is broken

### Detailed context

the wrist is stuck in a non linear position and it doesn't move

fbiggi, 2022-09-23:
insert the METERSPRINGS_T30900 springs in their seats
checked the tightening of the pins CER_009_P_073
Tripod calibration check is recommended
It seems that a joint has gone over the edge

[image]

I proceed with the reassembly of the wrist

fbiggi, 2022-09-23:
@ste93 
robot ready
after your check please close the issues

ste93, 2022-09-26:
Thank you @fbiggi , I will check the robot tomorrow.

ste93, 2022-09-27:
working perfectly, thank you @fbiggi

closed 2022-09-27 by ste93

fbiggi, 2022-09-28:
on 09/28/2022 the left forearm was reopened
the m1 engine was blocked in the downward stretch
perhaps the dirt was affecting the engine