robotology/mesh-support #1432: Mechanical backlash on right elbow

S-Dafarra, 2022-09-23:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right elbow has some mechanical slack.

### Detailed context

After some energetic handshakes during a demo, the right elbow started having some backlash.

GiacomoVassallo13, 2022-09-28:
Buongiorno Il gomito di iCub 3 presentava gioco. La causa derivante erano le viti della flexspain, che a causa del grasso abbondante e delle vibrazioni ha fatto allentare le viti.

[image]

L' assieme e' stato ripulito e su due delle viti in questione e' stato messo il frenafiletti. Durante l intervente e' stato cambiato anche un cuscinetto poiche rimasto danneggiato durante l estrazione e la flex dell Harmonic drive che si e' deformata durante l intervento.
L' assieme e' stato rimontato e testato manualmente, si attendono feedback in seguito alla movimentazione elettrica.
Nell intervento sono stato aiutato da @Elio75 e @emilianob80.

cc: @Fabrizio69 @S-Dafarra @DanielePucci @maggia80

gsisinna, 2022-09-30:
Elbow tested this morning and there seem to be no more of the problems mentioned above.

closed 2022-09-30 by gsisinna