robotology/mesh-support #1433: Initiation with ICUB

ALYPLY, 2022-09-23:
### Device name 🤖

iCubPlymouth01

### Request/Failure description

I am starting using this old ICub for the first time and followed the tutorial and installed the software. Then, I wanted to test the physical robot itself. So I followed the tutorial until I come to the RobotBazzar and when I choose an application and entered the name of the robot iCubPlymouth01, I found a message saying  (The selected robot has a pc104 processor. Therefore, Docker deployment can not be run on the head (the head node is disabled). And as most if the applications on the bazzar are related to the head, I am not able to test the physical robot.

### Detailed context

Considering the above:

1- How can I resolve this issue and test the physical robot?
2- I am not sure how would the robot move the joints while the joints are not in stiffness mode not to hurt the robot by any sudden movement, is this normal for ICub?
3- Is there any other recommended why to test the physical ICub, like connecting the simulator to the real robot?

gsisinna, 2022-09-26:
Hello @ALYPLY,
older robots with PC104 processors do not include the ability to run the docker deployment on the icub-head, but only on the workstation node. Usually these older robots used an NFS mechanism to mount the files from the icub-console to the head, so I would ask you to try to follow the official guide to install all the necessary:

[Superbuild installation on icub-head](https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_superbuild/)

and

[Generic iCub environment machines](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/generic-machine/)

[The iCub Console Server](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-laptop/)

Otherwise, iCub has a low-level control that provides hardware limits for each joint, so a minimum level of safety is already in place for joint movements but it is up to your control to prevent the joints from colliding with additional robot parts or objects in the scenario.

Finally, there is a Gazebo-based deployment (on RobotBazaar) that allows the robot to be tested on a simulator, and lately support for linking simulators to real robots is being improved (Blender Controller). Below are some references that might be useful to you:

[Gazebo Simulation Deployment](https://robot-bazaar.iit.it/applications/3)
[Blender Controller](https://robot-bazaar.iit.it/applications/29)

I remain available for any questions!

ALYPLY, 2022-10-04:
Hi 

Thanks for your support, I am trying to figure out, how to manage that through the links you sent. I have a question:

1- I have installed already robotology-superbuild following a previous installation (not the one corresponding to 104) and it is not installed by default in /usr/local/src/robot/robotology-superbuild as the directory /robot/ does not exist. So I had to create it and create a different robotology-superbuild directory. Should I remove the other version robotology-superbuild?

2- When I tried to compile robotology-superbuild following the tutorial link above you sent, there is no cmake file to run the cmake instruction.  so how can I manage this?

The version robotology-superbuild I checked out following the tutorial you sent me for 104 is 2022.09.0

Kindly advise

A

gsisinna, 2022-10-07:
Hi @ALYPLY,
in order:

1. I would use only one source of the robotology-superbuild (the one inside /usr/local/src/robot ). Be sure to import the correct path also in the .bashrc and .bashrc_iCub_superbuild files in your home directories.

2. To build the robotology-superbuild you must go inside the **build** directory and type this commands:

`ccmake ..`

Tick the flags you want to use (no gazebo).
Press two times `c` and then `g`.

After this: `make`

For the version it depends also on the Firmware you have on your iCub boards, here the full table for versioning: 

[Distro 2022.09.0](https://icub-tech-iit.github.io/documentation/sw_versioning_table/2022.09.0/)
[Firmware Versioning Table](https://github.com/robotology/icub-firmware-build#firmware-versioning-table)
[Detailed info v1.27](https://github.com/robotology/icub-firmware-build/blob/v1.27.0/info/firmware.info.xml)

closed 2022-12-22 by github-actions[bot]