robotology/mesh-support #1434: Left leg issues on ankle pitch and FT support

gabrielenava, 2022-09-26:
### Device name 🤖

iCubGenova04

### Request/Failure description

After a checkup of iCubGenova04 we noted that there are a couple of issues in the left leg: 

- the ankle pitch motor produces a strange noise when moving it by hand. Maybe there is something broken?

- there are loose screws in the FT support on the upper leg, and this causes a big backlash when moving the hip yaw joint

### Detailed context

cc @vpunithreddy @gsisinna 

### How does it affect you?

blocking iRonCub experiments

GiacomoVassallo13, 2022-10-04:
Buongiorno le viri che tengono il sensore FT sono state strette. Il gioco non sussiste piu. L intervento sulla caviglia come concpordato con @gabrielenava verra' svolto la prossima settimana quando il robot non sara' in uso cosi da operare con calma.

cc: @Fabrizio69 @gsisinna @AntonioConsilvio

GiacomoVassallo13, 2022-10-06:
Hi @gabrielenava could you tell me an ideal day to carry out the maintenance.

cc: @Fabrizio69 @DanielePucci

gabrielenava, 2022-10-06:
starting from Tuesday 11th any day is fine. I think it is better on Tuesday if possible so we have margin for other interventions (if needed) before we leave for the wind tunnel

gabrielenava, 2022-10-14:
thanks @Elio75 

we did not manage to test the robot in this week because we were using the setup to run experiments with turbines on the roof, but we'll test it on Monday to confirm that everything works!

gabrielenava, 2022-10-17:
we tested the robot today, everything looked fine! thanks again

Fabrizio69, 2022-10-18:
> we tested the robot today, everything looked fine! thanks again
... problem solved, we can close this issue

closed 2022-10-18 by Fabrizio69