robotology/mesh-support #1435: Right shoulder pitch motor issue

gabrielenava, 2022-09-26:
### Device name 🤖

iCubGenova04

### Request/Failure description

When trying to move manually the right shoulder joint, it looks very stiff compared to the other joints (and also the left shoulder pitch). The corresponding motor may be damaged, and we would need to check it.

side note: also the two elbow joints have a very different behavior when moving manually.

### Detailed context

cc @vpunithreddy @gsisinna @GiacomoVassallo13

GiacomoVassallo13, 2022-10-04:
Buongiorno, il gruppo motore del pitch della spalla destra e' stato smontato.
Fortunatamente, a differenza di quanto si supponeva il rotore non presentava danneggiamenti. I problemi erano causati dal cuscinetto callettato sull albero che si era consumato, producendo cosi atriti e sporco che si e' espanso nel motore. 

[image]

Il cuscinetto e' stato sostituito, purtoppo anche l encoder risultava danneggiato dallo sporco. Per questo  si e' deciso di sostituirlo. 
Durante il cablaggio e' stato necessario sostituire il pezzo cod. 1896 RIGHT PULLEY-CRIMP COVER poiche' la pressione dei cavi o ha portato a rottura.

ATTENDIAMO FEEDBACK.

cc: @Fabrizio69 @gsisinna @AntonioConsilvio

gabrielenava, 2022-10-04:
hello everyone, the robot started with no errors, but there is a calibration problem of the right shoulder encoder: the offset is big and the arm reaches the limits and goes in hardware fault while starting

cc @GiacomoVassallo13 @gsisinna

maggia80, 2022-10-04:
This is caused by the fact that the magnet could have changed its position wrt the AEA sensor after the operation describe above by Giacomo. The shoulder needs to be calibrated again.

gsisinna, 2022-10-05:
I will check this morning with @AntonioConsilvio the shoulder calibration

gsisinna, 2022-10-05:
The joint has been calibrated and tested with @gabrielenava and @AntonioConsilvio.

Closing this issue 👍

closed 2022-10-05 by gsisinna