robotology/mesh-support #1436: Several issues with iCub3 arms

S-Dafarra, 2022-09-28:
### Device name 🤖

iCubGenova09

### Request/Failure description

During the tests we encountered a set of issues on the arms, listed below.

### Detailed context

- The right wrist cables snapped
- The following fingertips are not working:
   - right thumb
   - right middle
   - right pinky
- The right thumb oppose/proximal were having delays while working in teleoperation, although they seem more or less fine from the motorgui. I wonder if some cables could be loose, or if there is an issue with the hall encoders.

maggia80, 2022-09-29:
cc @GiacomoVassallo13 @emilianob80

emilianob80, 2022-09-29:
@fbiggi

maggia80, 2022-09-29:
eyelids need to be recalibrated 

cc @davidetome , @salvi-mattia

salvi-mattia, 2022-09-29:
> eyelids need to be recalibrated
> 
> cc @davidetome , @salvi-mattia

as we said we should try to create sw limits slightly narrow than the hw ones

davidetome, 2022-09-29:
Calibration and limits [updated](https://github.com/ami-iit/robots-configuration/commit/311697ffc68bd48f0b5585a401377940242e1570), let's see the feedback (thanks @GiulioRomualdi 👍🏻 )

pattacini, 2022-09-29:
Thanks @davidetome 👍🏻

gsisinna, 2022-09-30:
In addition to the problems mentioned in the issue, this morning I did a deep check of the right arm, and several electrical problems were found, including a 2FOC with a blazed power channel for the shoulder yaw joint (joint 2), which was going into hardware fault. A connector for the wrist prosupination motor was redone and I checked all the right hand finger hall sensors, including the thumb which should now work better. The left wrist AEA harness was repaired, a CAN on the C-Shape was redone, and now calibration seems to be proceeding smoothly with all joints in green.
Only fingertips for which material has already been recovered should remain excluded.

cc @Fabrizio69 @maggia80

S-Dafarra, 2022-09-30:
Unfortunately, the left wrist keeps going in hardware fault.

GiulioRomualdi, 2022-10-03:
@gsisinna is on the task

gsisinna, 2022-10-05:
Have the above problems been observed again? The wrist should be ok 👍

cc @S-Dafarra

S-Dafarra, 2022-10-05:
The wrist seems ok now!

gsisinna, 2022-10-07:
Closing in favor of https://github.com/robotology/icub-tech-support/issues/1440

cc @S-Dafarra @Fabrizio69

closed 2022-10-07 by gsisinna