robotology/mesh-support #1437: Clusterwidget: port /pc104 not responding

pfdominey, 2022-10-03:
### Device name 🤖

iCubDijon01

### Request/Failure description

We just got the iCub back from Genoa.  Thanks
Two problems
1. using the new console, the computer fails to establish aconnection with the router.  Do you recall if this worked correctly in Genoa?
2. As a backup, we try to connect with the console we used before, which was fully reinstalled and updated at the n-1 trip to Genoa.  This is because there was a backlog with Dell, and you did not have our new console, so you updated our console.  This is what we were using up unitl now.

This computer connects to the network, we can ping to the pc104.  Then we start the yarpmanager,in the cluster tab of the gui with the yarpmanager, run the nameserver.  We try to start icub-srv and pc104 (see photo) and the pc104 gives an error ""Clusterwidget port pc104 is not responding" and the icon remains red.

Can you please advise on these problems?
thanks
Peter
[image]

### Detailed context

explained above:  
in the Cluster Management Window the pc104 does not connect, and the icon stays red.

IN the GUI window it says that that yarprun was successfully executed on the pc104, but in the cmd window, there is a failure, and the in the gui the icon remains red.  See attached screen shot.
[image]

pfdominey, 2022-10-03:
got the new console to connect by changing the network parameters as shown in the first photo

still have the problem with the pc104 icon staying red
[image]
[image]

pfdominey, 2022-10-04:
Hi @gsisinna 
Can you help with this?  We cannot use the robot.
I suspect that there is an ip address error somewhere.  Probably very simple to fix, but I dont know how to do it!!
thanks
cheers
Peter

valegagge, 2022-10-05:
Hi @pfdominey,

I suggest you do a simple test:
connect to pc104 by ssh from your laptop and then give the command:
`yarprun --server /pc104`.

In this way, we check that ssh is configured correctly and the yarp command works as expected.

Please report here the output of both commands. 
Thanks!
cheers 
Valentina

pattacini, 2022-10-05:
Hi @pfdominey 

While doing as @valegagge suggested, you may also check whether from the pc104 (right after the ssh) you can see the YARP network by running:
```console
yarp where
``` 

If the YARP server isn't detected, we will need to change the pc104 configuration.

Also, note down whether while ssh-ing into pc104 you're prompted with providing the password.

pfdominey, 2022-10-05:
hi!
Here's what happens:
icub@icubsrv:~$ ssh pc104
iCub LIVE image for PC104 version 7.10-21.04.27
Based on buster amd64
Build on 27-Apr-2021
Linux pc104 4.19.0-16-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.181-1 (2021-03-19) x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Oct  5 14:45:52 2022 from 10.0.0.1
Setting up yarp and iCub env vars
icub@pc104:~$ yarprun --server /pc104
-bash: yarprun: command not found
icub@pc104:~$

pfdominey, 2022-10-05:
so it seems that yarp is not running on the pc104?

pattacini, 2022-10-05:
Hi @pfdominey 

You should be using the laptop we sent you. There's no guarantee that the old one can be used alongside. Was that the case with the test?

>using the new console, the computer fails to establish aconnection with the router. Do you recall if this worked correctly in Genoa?

We don't provide routers. So, you can do the test by connecting the robot ethernet cable straight to the laptop console (or you can use a switch in the middle).

cc @gsisinna

pfdominey, 2022-10-05:
we are using the new laptop you sent us

pfdominey, 2022-10-05:
we are using the switch you provided

pfdominey, 2022-10-05:
I can ping to the pc104 and ssh to it, as seen in the screen output above

pfdominey, 2022-10-05:
but if I type "yarp" I get a response "yarp : command not found"

gsisinna, 2022-10-05:
hi @pfdominey can you try to follow [this](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-laptop/?h=ssh#install-the-ssh-keys-for-password-less-login-on-icub-head) steps for ssh keys? (with the new laptop)

it is really strange that you don't find yarp, and the new laptop sent was tested with the robot and worked.

Thanks, 
waiting for your feedback

pattacini, 2022-10-05:
>we are using the new laptop you sent us

Ok 👍🏻 

>we are using the switch you provided

This is strange; maybe it's a USB/ETH adapter instead?

>so it seems that yarp is not running on the pc104?

Most likely, YARP is there but not "exposed" somehow.

The output of the following commands may help:
1. `cat /etc/fstab` to be run on pc104
2. `echo ${PATH}` to be run on pc104

gsisinna, 2022-10-05:
Also @pfdominey , open a terminal in the new laptop (connected to the robot network) and type this command:

`ip a`

pfdominey, 2022-10-05:
icub@pc104:~$ cat /etc/fstab
overlay / overlay rw 0 0
tmpfs /tmp tmpfs nosuid,nodev 0 0
icub@pc104:~$

pfdominey, 2022-10-05:
icub@pc104:~$ echo ${PATH} 
/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
icub@pc104:~$

pfdominey, 2022-10-05:
> Also @pfdominey , open a terminal in the new laptop (connected to the robot network) and type this command: `ip a`

<pre><font color="#4E9A06"><b>icub@icubsrv</b></font>:<font color="#3465A4"><b>~</b></font>$ ip a
1: lo: &lt;LOOPBACK,UP,LOWER_UP&gt; mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp0s31f6: &lt;BROADCAST,MULTICAST,UP,LOWER_UP&gt; mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether b4:45:06:49:b8:b7 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.1/24 brd 10.0.0.255 scope global noprefixroute enp0s31f6
       valid_lft forever preferred_lft forever
    inet6 fe80::2607:a7b4:b501:c0/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlp0s20f3: &lt;BROADCAST,MULTICAST,UP,LOWER_UP&gt; mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether f8:b5:4d:fe:e2:87 brd ff:ff:ff:ff:ff:ff
    inet 192.168.43.101/24 brd 192.168.43.255 scope global dynamic noprefixroute wlp0s20f3
       valid_lft 2032sec preferred_lft 2032sec
    inet6 fe80::f089:d2c4:f5de:69af/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
<font color="#4E9A06"><b>icub@icubsrv</b></font>:<font color="#3465A4"><b>~</b></font>$ 
</pre>

pattacini, 2022-10-05:
>this is the router
black cable comes from from the rack and is from the pc104 and the grey cable connects to the console

I missed out the rack @pfdominey 👍🏻 
So, you have a switch yep.

Judging from the output you posted, it's very likely that PC104 is not mounting the partition that contains YARP, which in turn is provided by the server.

We would need to schedule a remote session eventually.
In the meantime, could post the output of the following command?
- `cat /etc/exports` (to be run on icubsrv)

pfdominey, 2022-10-05:
is it possible that the pc104 is not looking in the right place for the partition that is provide by the server?
I think I had to do a yarp --write at the beginning which might have changed an ini file or something?
any other things I could try?
thanks

pfdominey, 2022-10-05:
One other thing that may be important: at first the console could not connect to the switch.

I manually changed the network settings IPv4 to address 10.0.0.1 and dns automatic

could this cause a problem for the pc104?

pfdominey, 2022-10-05:
since everything was working in Genoa, it seems strange that its not working here, and probably something simple.

pfdominey, 2022-10-05:
Would you be available Monday afternoon for a video session and you could connect to the console by teamviewer and ssh to the pc104 and see whats wrong?

pattacini, 2022-10-05:
>```
>/exports/code 10.0.0.0/255.255.255.0(rw,sync,no_subtree_check)
>/exports/local_yarp 10.0.0.0/255.255.255.0(rw,sync,no_subtree_check)
>```

The server does correctly export the YARP code although the PC104 is unable to mount it.

>is it possible that the pc104 is not looking in the right place for the partition that is provide by the server?

That's the most probable hypothesis yep

>I think I had to do a yarp --write at the beginning which might have changed an ini file or something?

If `yarp --write` worked then it means that you found YARP. Likely, you launched it on the server. However, here the problem seems to be the file system, thus not the YARP network. At any rate, `yarp -write` is required from time to time when the underlying ethernet is changed.

>I manually changed the network settings IPv4 to address 10.0.0.1 and dns automatic

This is actually particularly weird as the laptop should have been already configured with the proper options. Anyway, the settings you used look good as the PC104 expects to mount from and talk to 10.0.0.1 (although the content of `/etc/fstab` was incorrect).

>Would you be available Monday afternoon for a video session and you could connect to the console by teamviewer and ssh to the pc104 and see whats wrong?

I think that we can organize the session. Let me verify with our IT manager. Most presumably, we will be available as our internal tests will be over. Stay tuned, I will confirm this.

pattacini, 2022-10-06:
Hi @pfdominey 

I can confirm we can schedule a remote session on Monday 10 in the afternoon.
@davidetome and @davidelasagna will be dealing with it.

Does the slot 14:30-16:00 work for you?
If so, I can email you all the invitation.

pfdominey, 2022-10-10:
Hi - Yest this slot works.
I am ready now if you want :)

pfdominey, 2022-10-10:
Thanks for working on this strange problem.
I am available any time tuesday and wednsday to try again by teamviewer.
cheers
Peter

pfdominey, 2022-10-10:
last words for the evening!
Hope to see you tomorrow

Scanning dependencies of target iDyn
[  8%] Building CXX object src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDyn.cpp.o
[  8%] Building CXX object src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDynInv.cpp.o
[  8%] Building CXX object src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDynBody.cpp.o
[  8%] Building CXX object src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDynTransform.cpp.o
Assembler messages:
Fatal error: can't create CMakeFiles/iDyn.dir/src/iDynTransform.cpp.o: Input/output error
make[5]: *** [src/libraries/iDyn/CMakeFiles/iDyn.dir/build.make:102: src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDynTransform.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:2857: src/libraries/iDyn/CMakeFiles/iDyn.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/ICUB.dir/build.make:109: src/ICUB/CMakeFiles/YCMStamp/ICUB-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1298: CMakeFiles/ICUB.dir/all] Error 2
make: *** [Makefile:95: all] Error 2
icub@pc104:/usr/local/src/robot/robotology-superbuild/build-pc104$

pattacini, 2022-10-10:
Hi @pfdominey 

We'll get back to you asap!
Updating the dependencies resulted in some progress at least.

Now the problem is:
```
Assembler messages:
Fatal error: can't create CMakeFiles/iDyn.dir/src/iDynTransform.cpp.o: Input/output error
```

It seems related to the disk space, somehow.
Tomorrow, could you please run the following on PC104?
```console
df -h
```
reporting here the outcome.

Thanks!

pfdominey, 2022-10-10:
Will do!  I will do it around 8h45 or so.
thanks again!
cheers

pfdominey, 2022-10-11:
Let me know when you want me to connect to the pc104 and teamviewer :)

pattacini, 2022-10-12:
Hi @pfdominey 

After some struggling, we finally managed to make the robot start up correctly.

The remaining issues are related to the issue of `ctpService`.
It would help us a lot to have the following material:
- A video of the robot failing while controller with `ctpService`.
- The XML file launching the `ctpService` services.
- The bash script with the calls to `ctpService`.
- The log of the `yarprobotinterface` while the robot fails.

@gsisinna could assist in this.

pfdominey, 2022-10-12:
Thanks for the continued efforts!

For ctpServices, I launch 4 seperate consoles and make these commands, respectively in the 4:

ctpService --robot icub --part right_arm
ctpService --robot icub --part left_arm
ctpService --robot icub --part head
ctpService --robot icub --part torso

Once these are running I then use the program:
protocol-balanced_fixing.sh
I send the .sh file by email because .sh is not allowed

from a console I execute:
./protocol-balanced_fixing.sh home
this works ok,  then I execute
./protocol-balanced_fixing.sh point6Right
in the video, we see this fail, with the wrist snapping down

[yarprunlog_12_10_2022_14_53_15-wrist-fail.log](https://github.com/robotology/icub-tech-support/files/9765039/yarprunlog_12_10_2022_14_53_15-wrist-fail.log)

pfdominey, 2022-10-12:
more info:  here is the video of the joint failure

then in the second video, the wrist should be in the same position ("home" in the .sh script), but we see that it has fallen down

https://user-images.githubusercontent.com/32130758/195350253-10f3b6e3-6886-4b9b-8b44-ff600be16fd6.mp4

https://user-images.githubusercontent.com/32130758/195350594-234261dd-8636-4573-a204-4e64a770f7d6.mp4

pattacini, 2022-10-12:
Thanks heaps @pfdominey for the useful info!

>I send the .sh file by email because .sh is not allowed

You can zip it and enclose in the issue.

pfdominey, 2022-10-12:
the script is a bit messy.  what we use are the functions  seqA-seqJ, home, home_look, home_down, calibrate

each of he sequences of pointing movements seqA-seqJ each has 10 seperate pointings

pfdominey, 2022-10-12:
can you see the videos?

gsisinna, 2022-10-12:
> can you see the videos?

No @pfdominey, there no videos uploaded.

Thanks!

pattacini, 2022-10-12:
>can you see the videos?

Yep, thanks 👍🏻 

Could you please try the following to be sure that we can log everything relevant?

On the PC104, launch the `yarprobotinterface` with the magic below that serves to redirect the output to a file:

```console
yarprobotinterface > output.log 2>&1
```

Finally, post here the file `output.log` produced.

pfdominey, 2022-10-12:
ok you want me to run yarprobotinterface from the pc104 with the log file, and then duplicate the problem again?

gsisinna, 2022-10-12:
Yes @pfdominey, by replicating the problem and extracting the log we would have all the useful information about it. Thanks

pfdominey, 2022-10-12:
in this case, the wrist failed, fell down,  at the end of the calibration.  We did not have to use ctpServices
the 4 ctp services were not running.
I do not know how to transfer the log file from the pc104.  

can you tell me how to do that?

pfdominey, 2022-10-12:
in the mean time I am going to shet everything off and then try again to make sure it was not due to an undefined state

pattacini, 2022-10-12:
The latest failure may be due to the fact that you launched the yarprobotinterface not from the intended path.

Do this then:

```console
ssh pc104
cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01
yarprobotinterface > output.log 2>&1
```

Afterwards, you should be able to retrieve the file `output.log` straight from the laptop from the following path:
```console
/exports/code/robotology-superbuild/src/robots-configuration/iCubDijon01
```

pfdominey, 2022-10-12:
ok - when starting the robot from this directory, the wrist does not fail
now using ctpservices
the first use of ctpservice to move to our first (home) position causes the wrist to fall
[third-log-file.log](https://github.com/robotology/icub-tech-support/files/9765901/third-log-file.log)

pfdominey, 2022-10-12:
here is a video where we have done this
ssh pc104
cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01
yarprobotinterface > output.log 2>&1

the robot startup is good.  We then exeute this command
icub@icubsrv:~/pointing_sequences$ ./protocol-balanced_fixing.sh home
and at the wrist fails

https://user-images.githubusercontent.com/32130758/195372530-9c828772-a6aa-4c1e-bc07-620f74fb6332.mp4

pfdominey, 2022-10-12:
in case the video is not visible, here is a link
https://youtu.be/1AlHn0kfhbs

pattacini, 2022-10-12:
Do you have the `output.log` to share here?

pfdominey, 2022-10-12:
it is above
https://github.com/robotology/icub-tech-support/files/9765901/third-log-file.log

pfdominey, 2022-10-13:
Experiment 1 - starting the robot from yarpmanager gui
The wrist 'drops' during the calibration process - no need to invoke ctpservice
[yarprunlog_13_10_2022_10_15_14.log](https://github.com/robotology/icub-tech-support/files/9772382/yarprunlog_13_10_2022_10_15_14.log)

pfdominey, 2022-10-14:
We just did testing with @gsisinna and everything is working
apparently there was only a problem with the red icon in the gui
there was the message
yarprun successfully executed on pc104

we then tested an updated set of movements that solicity joint 6 with smaller amplitudes, and we checked that movements were correct, and motorgui showed correct results

everything is working!  I will now proceed with some final testing before restarting our experiments next week.  After that I can close the issue

gsisinna, 2022-10-14:
Thank you @pfdominey, I am glad that now the movements are going better and it seems to me that from the modification of the script you use for your tests it is going much better!

pfdominey, 2022-10-14:
I can now happily say that we did extensive testing, did a restart of everything, and ...... the problem is closed!
Thanks @gsisinna and everyone on the team in Genoa!
We will start our experiments again on Monday!

closed 2022-10-14 by pfdominey

pattacini, 2022-10-14:
Glad that iCubDijon01 is back operative!

Thanks @davidelasagna @davidetome @gsisinna @Nicogene @traversaro!