robotology/mesh-support #1439: iCubGenova01 - Left hand index distal motor is broken

xEnVrE, 2022-10-05:
### Device name 🤖

iCubGenova01

### Request/Failure description

While investigating on https://github.com/robotology/icub-tech-support/issues/1396, @gsisinna notified us about a possible broken motor of the left hand index distal joint.

### Detailed context

While investigating on https://github.com/robotology/icub-tech-support/issues/1396, @gsisinna notified us about a possible broken motor of the left hand index distal joint.

xEnVrE, 2022-10-24:
Last week the motor has been replaced, I think by @GiacomoVassallo13.

Unfortunately, at the moment there is something strange going on at calibration time with the left arm. The calibration does not complete and, after a while, the robot self-shutdowns itself.

I am attaching here the log from the `yarprobotinterface` to ease the investigation.

[log_icub-head_yarprobotinterface_1295.txt](https://github.com/robotology/icub-tech-support/files/9849899/log_icub-head_yarprobotinterface_1295.txt)

For now, we disabled the entire left arm so that the robot can be used with the right arm only.

GiacomoVassallo13, 2022-10-24:
Buongiorno. Il vecchio motore dell indice distale e' stato sostituito.Il motore e' stato attaccato sul supporto e montato sull avanbraccio. Il tendine e' stato ripristinato correttamente e non sembrano esserci problemi di natura meccanica. adesso con @AntonioConsilvio andremo a controllre i problemi evidenziati nel commento di @xEnVrE ( https://github.com/robotology/icub-tech-support/issues/1439#issuecomment-1288593980 commento qua sopra). 

cc: @Fabrizio69

Fabrizio69, 2022-12-14:
iCubGenova01 will be dismissed soon, hence closing.

closed 2022-12-14 by Fabrizio69