robotology/mesh-support #1440: Right hand not working properly

S-Dafarra, 2022-10-06:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right hand of the robot is having several issues

### Detailed context

Here the issues:
- index distal broken
- middle distal nearly broken
- thumb oppose in a bad shape
- thumb oppose hall sensor not working

paolo-viceconte, 2022-10-06:
After today's integration tests, the following issues affect the robot:

1. The skin of the right index, middle, ring and pinky fingers does not work
2. The right wrist prosup and thumb proximal were often in HF due to hard limit reached
3. The robot battery does not recharge
4. The robot battery switch is not fixed to the cover and goes out while the robot walks

cc @DanielePucci @pattacini @S-Dafarra @GiulioRomualdi

S-Dafarra, 2022-10-06:
> 2\. The right wrist prosup and thumb proximal were often in HF due to hard limit reached

This was probably due to the encoder, because it was possible to recover them afterwards

emilianob80, 2022-10-07:
> > 2. The right wrist prosup and thumb proximal were often in HF due to hard limit reached
> 
> This was probably due to the encoder, because it was possible to recover them afterwards

the motor has been tested with the jogger and presents no problems. The encoder reads correctly

cc: @maggia80

emilianob80, 2022-10-07:
> After today's integration tests, the following issues affect the robot:
> 
> 1. The skin of the right index, middle, ring and pinky fingers does not work
**T.B.D.**

> 2. The right wrist prosup and thumb proximal were often in HF due to hard limit reached
**wrist motor and the encoder are working correctly. The proximal thumb problem has been identified as mechanical (@GiacomoVassallo13 is working on it). It appears to be a loose tendon problem and therefore functioning is impaired

> 3. The robot battery does not recharge
**fixed**

> 4. The robot battery switch is not fixed to the cover and goes out while the robot walks
> **fixed**

> cc @DanielePucci @pattacini @S-Dafarra @GiulioRomualdi @maggia80 @Fabrizio69

GiacomoVassallo13, 2022-10-07:
Hi, all the joints of right hand have been reworked. The proximal tendon windings have been ordered and stretched. No cables were replaced, as no damage was found. I am now going to check the left hand.

cc: @DanielePucci @pattacini @S-Dafarra @GiulioRomualdi @maggia80 @Fabrizio69

pattacini, 2022-10-08:
The right hand is now working again.

closed 2022-10-08 by pattacini