robotology/mesh-support #1441: Battery seems not charging

DanielePucci, 2022-10-06:
### Device name 🤖

iCubGenova09

### Request/Failure description

After some tests for the ANA Avatar XPRIZE, the battery seems not to charge (emitted current equal to zero) while the battery level is about 30%

### Detailed context

None

isorrentino, 2022-10-06:
As soon as I turned off the motors and the head, I tried again and it starts charging. I don't know if this can help to understand what is going on.

emilianob80, 2022-10-07:
The battery does not actually charge either normally or with the "wake-up" charge. For this reason it was decided to replace it with the new one in stock and investigate the problem on that one later.

cc: @maggia80

emilianob80, 2022-10-07:
La batteria nuova carica

cc: @maggia80 @DanielePucci

closed 2022-10-07 by emilianob80

emilianob80, 2022-10-07:
La vecchia batteria é sotto test e sembra funzionare anche lei ora