robotology/mesh-support #1442: Issues with the left wrist prosup and yaw

paolo-viceconte, 2022-10-07:
### Device name 🤖

iCubGenova09

### Request/Failure description

During today's integration test, the left wrist prosup and yaw went in HF. The encoder seemed not to be working.

### Detailed context

Please check above

### Additional context

cc @DanielePucci @S-Dafarra @GiulioRomualdi

maggia80, 2022-10-07:
with @salvi-mattia we were discussing the possibility that the motor is sliding when moving versus the hard-stop.

[image]

@emilianob80 with the jogger you could move the prosup vs the hard limit and keep pushing the motor. If the encoder keeps counting we know the reason.

we need to tight the two grub screws.

[image]

emilianob80, 2022-10-08:
@maggia80 The wrist prosup joint does not slip mechanically and works correctly with the jogger. The yaw joint of the wrist is functioning properly. I have tested the AEA sensor and it reads regularly. I have encountered a problem with the 4N2 cable connecting to the MC4Plus board which controls the AEA wrist sensors. I think that the cable, being engraved, created problems for the board and consequently for the sensors connected to it

cc: @pattacini @DanielePucci @S-Dafarra

closed 2022-10-08 by pattacini