robotology/mesh-support #1445: Left wrist yaw encoder issues

S-Dafarra, 2022-10-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

After some use, the left wrist encoder yaw indicates more than 300degs

### Detailed context

We were doing some gentle motions while holding a light bottle.After a while, the encoder started reading strange values and the joint went into hardware fault

cc @DanielePucci @isorrentino @pattacini @maggia80 @emilianob80

emilianob80, 2022-10-09:
@S-Dafarra on Monday morning I will wire a sensor on the bench in order to replace it completely on Monday or, if there is no time, in Los Angeles as soon as the robot is available.

cc: @DanielePucci @maggia80

S-Dafarra, 2022-10-09:
Just to be sure, were you referring to this issue, or to https://github.com/robotology/icub-tech-support/issues/1446?

emilianob80, 2022-10-09:
> Just to be sure, were you referring to this issue, or to #1446?

I was referring to the problem of the encoder pulse (subject of the issue). Regarding the FT sensor problem comment in the reference question

cc: @pattacini @maggia80 @DanielePucci

maggia80, 2022-10-09:
the value read by the AEA encoder could go from 0 to 360 degrees. If it reads a number that is in this range it's quite probable that is reading correctly. 
We should check if the magnet is blocked or if it's loose and rotates.

maggia80, 2022-10-10:
the magnet was rotating inside its housing. @Gandoo glues it properly. 

We need to calibrate the left wrist yaw once in LA.

cc: @davidetome

davidetome, 2022-10-10:
> the magnet was rotating inside its housing. @Gandoo glued it properly.
> 
> We need to calibrate the left wrist yaw once in LA.
> 
> cc: @davidetome

I guess we also need to adjust its position before calibrating the joint, am I right @maggia80 ?

maggia80, 2022-10-10:
> > the magnet was rotating inside its housing. @Gandoo glued it properly.
> > We need to calibrate the left wrist yaw once in LA.
> > cc: @davidetome
> 
> I guess we also need to adjust its position before calibrating the joint, am I right @maggia80 ?

no, the wrist yaw it's read by an AEA and we don't need to adjust it.

davidetome, 2022-10-10:
> no, the wrist yaw it's read by an AEA and we don't need to adjust it.

yep! I was confusing w/ the fingers

closed 2022-11-06 by pattacini