robotology/mesh-support #1446: Left arm FT sometimes reads all zeros

S-Dafarra, 2022-10-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

There seem to be some wiring issue on the left arm and sometimes the readings are all zero.

### Detailed context

The FT reads some sporadic zero from time to time

cc @isorrentino @DanielePucci @pattacini @maggia80 @emilianob80

emilianob80, 2022-10-09:
@S-Dafarra Regarding the FT sensor problem, as agreed with @ugo verbally, tomorrow morning while I remove the battery and then open the covers, I check the wiring on the chest boards. If you do not find anything, however, I wire a cable from the EMS4 card of the chest to the FT sensor in order to be able to completely replace it in case of need or if it is explicitly requested.

cc: @maggia80 @DanielePucci

maggia80, 2022-10-09:
@S-Dafarra you get zeros from the yarp port, or you don't get any value for a while?
Because if you get some zeros, it means that the CAN messages and then the ETH messages are sending these numbers. 
Therefore the issue is on the STRAIN2 board and not in the CAN connection. Alternatively, if you don't get any message there might be some disconnection from the CAN bus and @emilianob80 intervention is effective.

S-Dafarra, 2022-10-09:
We read zeros from the yarp network.

If the CAN reading is always zero, can it be because the value was somehow corrupted, or even in that case we should see some number?

maggia80, 2022-10-09:
you should see some numbers. The CAN message cannot get corrupted with all number reading zeros. It could lost packets.
But zero means the output of the sensor is zero. Some saturation at the level of the FT or some short to GND of the signals inside the FT.

emilianob80, 2022-10-10:
@S-Dafarra After removing the covers, I checked the 2N2 cable that connects the EB2_P5 board with the Left Upper Arm F / T sensor. The cable is in excellent condition. However, it will be my responsibility to prepare one to replace if necessary

cc: @maggia80 @DanielePucci @pattacini

pattacini, 2022-10-10:
Hi @emilianob80 
Probably, it is convenient to leave this issue open as a reminder for LA.
What do you think?

pattacini, 2022-10-10:
Discussed T2T and we agreed to open it up again.

closed 2022-11-06 by pattacini