robotology/mesh-support #1447: Left pinky does not calibrate

S-Dafarra, 2022-10-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

The left pinky finger does not calibrate properly

### Detailed context

During calibration, the left pinky closes but it does not open.

maggia80, 2022-10-09:
what's the value of joint position after calibration? 
Could you move the pinky after calibration if you enable it from the motorgui?

S-Dafarra, 2022-10-09:
> what's the value of joint position after calibration? Could you move the pinky after calibration if you enable it from the motorgui?

It was not calibrated. We did not check the value, but it was not crazy. Now it seems to calibrate, so probably it is just slightly miscalibrated

emilianob80, 2022-10-10:
It seems unnecessary to intervene on the HW. Do I consider this issue closed @S-Dafarra  @maggia80 ?

cc: @DanielePucci  @pattacini

S-Dafarra, 2022-10-10:
Probably it is just matter of tweaking the joint calibration

closed 2022-11-06 by pattacini