robotology/mesh-support #1448: Issue with network on the head

S-Dafarra, 2022-10-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

When moving the neck, the ethernet connection of the robot has issues.

### Detailed context

We noticed that the forces explode when moving the head, indicating some issues in the communication.

See the video 

https://user-images.githubusercontent.com/18591940/194762906-5237bf76-e424-44a6-8fb6-ff3b5a98983f.mp4

After that, the robot stopped working. Here the log of the last messages (it was saturated before)

[log_icub-head_yarprobotinterface_280727_ethernet_issue.txt](https://github.com/robotology/icub-tech-support/files/9741807/log_icub-head_yarprobotinterface_280727_ethernet_issue.txt)

S-Dafarra, 2022-10-09:
The robot did not start for a while with the error mentioned in the log.

Then @davidetome showed up and it started working again.

(full disclosure, we did not do anything. There is likely an intermittent issue on the ETH line)

maggia80, 2022-10-09:
the failure seems at the level of the first connection from the com express type 10 and the MC4-PLUS in the head EB20 (10.0.1.20). 
We could check the cable E0.

davidetome, 2022-10-09:
> the failure seems at the level of the first connection from the com express type 10 and the MC4-PLUS in the head EB20 (10.0.1.20). 
> We could check the cable E0.

Ciao @maggia80 .. actually It seems there were some delay in the network.. Also opening Simply a file with gedit It took few seconda..

emilianob80, 2022-10-10:
@S-Dafarra The problem has been fixed. It was necessary to re-crimp connector E0 on the ComExp electronic board. Network problems should be fixed

cc: @maggia80 @DanielePucci @pattacini @davidetome

pattacini, 2022-10-10:
Probably, also this issue could stay open as a reminder for us when we'll be in LA to check at the first robot start-up.

pattacini, 2022-10-10:
Just put in review.

emilianob80, 2022-10-10:
Sorry @pattacini , this morning the robot was tested with @S-Dafarra . Is it worth reopening the issue?

pattacini, 2022-10-10:
> Sorry @pattacini , this morning the robot was tested with @S-Dafarra . Is it worth reopening the issue?

Ok, didn't know that.
Closing then.

closed 2022-10-10 by pattacini