robotology/mesh-support #1451: Left wrist behaving not correctly

DanielePucci, 2022-10-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

Unfortunately we just noticed that the left wrist does not follow the reference positions, although is up and running. It seems like it slides internally

### Detailed context

As above

emilianob80, 2022-10-10:
@S-Dafarra The problem seems to be due to the magnet which was not glued in its seat and therefore moved in certain operations. Problem fixed. Tested with the jogger and working

cc: @maggia80 @DanielePucci @gaurvigoyal

closed 2022-10-10 by emilianob80

Gandoo, 2022-10-10:
on the carbon palm the spacers are broken, fixed and reassembled after gluing the magnet

[image]

[image]

cc: @maggia80 @DanielePucci @gaurvigoyal