robotology/mesh-support #1453: Crack in the robot torso metal support

gabrielenava, 2022-10-19:
### Device name 🤖

iCubGenova04

### Request/Failure description

after inspecting the robot with @GiacomoVassallo13 we noted a pretty scaring crack in the torso support:

[image]
 

### Detailed context

it would be nice to schedule an intervention to fix/replace the piece before the robot splits into two during an iRonCub experiment, although it would be amazing to see it. 

If the intervention takes too much time we can schedule it after the wind tunnel experiments of 3-4 november

Fabrizio69, 2022-10-20:
@gabrielenava this intervention is foreseen after the wind tunnel, when a spare part (code 3916 - RC_USAL_001_P_116) will be available.

cc: @GiacomoVassallo13 @AntonioConsilvio @maggia80

GiacomoVassallo13, 2022-12-14:
Hello. the interventions on the robot are concluded. The upperbody has been separated from the lowerbody so as to reach the affected piece.
Once this was done, the components from the old piece were recovered and installed on the new one. The new part was reassembled and the robot reattached. The tendon of the joint has been rewired.
@gabrielenava you can get the robot back.

cc: @Fabrizio69

gabrielenava, 2022-12-19:
Tested today, there were issues

We were not able to start the robot. The yarprobotinterface crashed with this error (this was the first error, then many others occur. See the complete logger)

```
 0,824503 <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD torso-eb5-j0_2 IP 10.0.1.5 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion
```

One time we managed to start the robot, but after few minutes the yarpmotorgui crashed and we were not able to recover it.

cc @AntonioConsilvio @Fabrizio69 @GiacomoVassallo13 

[yarprunlog_19_12_2022_18_40_47.txt](https://github.com/robotology/icub-tech-support/files/10261584/yarprunlog_19_12_2022_18_40_47.txt)
[log_icub-head_yarprobotinterface_1251.txt](https://github.com/robotology/icub-tech-support/files/10261586/log_icub-head_yarprobotinterface_1251.txt)

maggia80, 2022-12-20:
@AntonioConsilvio I will check if the board is visible from FirmwareUpdater.

AntonioConsilvio, 2022-12-21:
Hi @gabrielenava, I checked on 'FirmwareUpdater' the boards, and I could only see 4 boards. 

So I realized that the problem was in the ethernet cable that connect EMS called EB1 to the EMS called EB3 (because there aren't MC4-plus boards on ironCub)

To fixed the problem I just removed the covers and pressed the connector better, nothing more. 

I don't know if it will be a resolution or a final resolution. I tested the robot several times and it works.

I await feedback on whether or not the problem has been solved.

Instead, regarding torso, I tried it and it works! @GiacomoVassallo13 

@Fabrizio69 @maggia80

gabrielenava, 2022-12-22:
I have used the robot for the whole afternoon yesterday, and I did not have any problem. I think we can close the issue! I'll open a new one if the error appears again, thanks!

closed 2022-12-22 by maggia80