robotology/mesh-support #1454: Left arm issue with the forearm

ddetommaso, 2022-10-25:
### Device name 🤖

iCubErzelli02

### Request/Failure description

During calibration the forearm of the left arm does not go in the correct position.

[log_icub-head_yarprobotinterface_3327.txt](https://github.com/robotology/icub-tech-support/files/9858608/log_icub-head_yarprobotinterface_3327.txt)

https://user-images.githubusercontent.com/3440640/197730940-2c2d4b72-7281-4140-b8b6-b8b391a242bd.mp4

### Detailed context

no more details

ddetommaso, 2022-10-25:
https://user-images.githubusercontent.com/3440640/197732734-26d0b84b-0cad-44f4-b03e-52577921924f.mov

AntonioConsilvio, 2022-11-04:
Hello @ddetommaso, the problem about your issue ([#1454](https://github.com/robotology/icub-tech-support/issues/1454)) was about the connection between the motor **1B2M0 ( joint 4 )**, and the board **MC4_PLUS (EB24)**. Because the connector was quite detached.

Once it was fixed, we found an other problem during the closing of the cover. When the `yarprobotinterface` started, the terminal show us a strange hardware fault status on joint 3, followed by this message: 
```bash
2FOC hall sensors fault. Invalid sequence in notor Hall effect sensors, please check motor hall cable connections. part6= ID of joint. + .
```

So I checked the cable connections of the motor **1B1M1** ( **joint 3** ), and then I found the problem between the connection of cable **1B1H1** and the cable **1B1_P1**, in fact the **ground (GND)** at **pin 6** wasn't connected. 

After fixing this connection, the robot started to work well. 

Thank you for your help during this work.

cc @Fabrizio69 @maggia80

AntonioConsilvio, 2022-11-04:
`yarprobotinterface` video

https://user-images.githubusercontent.com/114915464/200009251-aba92cd6-a496-4dfc-8c96-06a37fde1a12.mp4