robotology/mesh-support #1456: Right forearm arm failure and skin problems

rozliv, 2022-11-07:
### Device name 🤖

iCubPrague01

### Request/Failure description

Right forearm and wrist joints are not working (in HW fault).
Moreover torso skin and right hand skin are not working and right forearm and upperarm skin have many phantom activations. 
@matejhof

### Detailed context

While using, the right wrist fell into HW fault; after a restart, neither the skin nor the joints are working. 
The torso skin probably has not been working for a longer time. 

[image]
[image]
[image]
[image]

AntonioConsilvio, 2022-11-14:
Hello @rozliv , we received your issue and we will contact you soon with further updates.

cc @Fabrizio69

matejhof, 2022-11-21:
Hello, 
is there any progress on this please? We have a timeline to run experiments...
Thanks!

AntonioConsilvio, 2022-11-22:
Hi @matejhof and @rozliv I have some questions to better understand the problems with your robot.

The first one is: the parts of the robot that fell into HW error, if you try to move these joints manually with `yarpmotorgui` open (if you can put these joints in idle it should be better but it is not mandatory), does the encoder value change? Or is it always the same?

Another thing, you should use the `FirmwareUpdater` command on the iCub head (with -X because it is a graphical application), then select the first option and click Discover. Here you will be able to see all the electronic boards in your robot, so send us a screenshot of what you see.

If you have questions about either of these, write to us.

cc @Fabrizio69

rozliv, 2022-11-22:
Hi @AntonioConsilvio, thank you for the response.
Encoders for joints 5 and 6 change values while moving, encoders for joints 4 and 7 change values even without moving (now the encoder for Joint 7 show around -2.4 million degrees and it is still decreasing. 

[image]

AntonioConsilvio, 2022-11-22:
Hi @rozliv, I need an other information. If you run `yarprobotinterface` on icub head from a bash, you should see several errors about HW fault status of joints 4,5,6 and 7 (and others error if there are any). Can you send us a screenshot of all these errors?

Then you have to do the same thing you did before with `FirmwareUpdater`, but after clicking on Discover, you should select the board mc4plus with address 10.0.1.22. 
After that, click on Force ETH Maintenance. Then (always with the board 10.0.1.22) click on Discover. Here we can see if the head can see the skin boards. So send us a screenshot. 

After these steps, restart the boards.

cc @Fabrizio69

AntonioConsilvio, 2022-11-29:
Hi @rozliv in FirmwareUpdater we can see that the skin boards are not visible. So probably a cable that connects mc4plus to MTB4 ( the skin board) came off.  

Instead regarding joints 5 and 6 maybe they just need a recalibration, the strange thing is for joints 4 and 7 that keep changing value without touching them. 

I think if joint 7 is -2.4 million degrees and still decreasing it probably has a large variation in its value (I don't know why yet). But what is the behavior of joint 4? Does it keep decreasing or does it do other things?

cc @Fabrizio69

rozliv, 2022-11-29:
Hi @AntonioConsilvio I am enclosing a photo of joint 4 after startup procedure and video of weird joint 4 movements afterwards @matejhof [image]

https://user-images.githubusercontent.com/45361823/204492114-f870bff5-7e7c-47c2-8543-2c865dfa73d1.MOV

AntonioConsilvio, 2022-11-29:
Thanks for the video @rozliv , I need to see photos of the AEA sensor of joint 7 too. This is between pinky and ring fingers. I need these views (maybe with flash is better), I colored what I need to see. Because we need to see whether the sensor is well placed or not. 

Front view:
[image]

This viewpoint is to see if it is aligned in its seat:

[image]

Last from below:
[image]

cc @Fabrizio69

AntonioConsilvio, 2022-11-29:
Whereas regarding joint 4, so if the motor makes this strange movement, on `yarpmotorgui` you should see that the encoder value fluctuates between two values that should be about equal in time (this pair of values does not change), is that correct? 

cc @rozliv @Fabrizio69

matejhof, 2022-11-29:
I took these two pictures now. Does it help? @AntonioConsilvio 
[image]
[image]

matejhof, 2022-11-29:
Now I started the robot. It came up with all green but joint 4. 
[image]

AntonioConsilvio, 2022-11-30:
This situation is much better than before. But is it just a case or does the robot work better now?
Because if this is the situation maybe it would be enough to recalibrate joint 7 (if the steel tendon is not broken, but it doesn't seem from the photos).

Joints 5 and 6 are in idle because joint 7 is in HW error; if in this situation you press run and then home on joints 5 and 6, they should return to position and run smoothly.

So, how does the robot work now? How in this photo?

cc @Fabrizio69

matejhof, 2022-11-30:
Thanks @AntonioConsilvio 
I can get to this state by running and homing joints 5 and 6.
[image]
@rozliv  says that joint 4 would get into HW fault only after some time. Yesterday I had this weird oscillations at 30 deg position. This is not happening right now, but something must still be strange there. 
Like yesterday, torso and right hand skin (palm and fingertips) are not responding at all.

AntonioConsilvio, 2022-12-01:
Hi @matejhof @rozliv, regarding the skins I think there is a disconnected (or almost disconnected) electrical cable so the CPU can't see the skin board (MTB4) on the torso and probably is the same for the other body parts. 

But let's analyze step by step, so now what is your priority, because joint 7 is often less important than others, so it can wait (if you don't need it). 

But joints 4, 5 and 6 give other problems? 

Maybe you can use the robot several times and write in this issue all the times these joints fell HW fault, to figure out how often you have problems with these joints (especially with joint 4) and decide if it is an important problem or if this problem occurs once in ten times. 

If the situation is like [this](https://github.com/robotology/icub-tech-support/issues/1456#issuecomment-1332196941) and joint 4 give no problems, what is your priority? Skins or Joint 7?  

But first we need to understand if joint 4 is really working well or not.

cc @Fabrizio69

matejhof, 2022-12-01:
We're trying to do some diagnostics here. As for the skin-related connections that could be loose / disconnected, where should we look for them @AntonioConsilvio ?

matejhof, 2022-12-01:
Hi @AntonioConsilvio, 
we're making progress here. The source of some of the problems was the jumper corresponding to the ring finger. In fact, the finger was hot. 
[image]
After we removed the jumper, all joints are working fine. Also the skin of the right hand is fine. When putting the jumper back, we're back to the errors we had before.

What remains dead is the torso skin. In addition, we realized now that also all the skin on the legs is dead (we did not check this before).
cc @0xBEDA @chatom98 @Fabrizio69

AntonioConsilvio, 2022-12-02:
> Hi @AntonioConsilvio, we're making progress here. The source of some of the problems was the jumper corresponding to the ring finger. In fact, the finger was hot. [image] After we removed the jumper, all joints are working fine. Also the skin of the right hand is fine. When putting the jumper back, we're back to the errors we had before.
> 
> What remains dead is the torso skin. cc @0xBEDA @chatom98 @Fabrizio69

Hi @matejhof @rozliv, when you remove this connector you disconnect the electronic board on fingertip (FTL2).
[image]

So, to understand why it makes this problem, you should check that the cable isn't broken
[image]

And if it is ok, try to remove the nail with this screw
[image]

**!WARNING!** You must be very very delicate, because these cables are so delicate. 

After removing it, you can check the soldering of the cables. Send us a photo of what you can see

cc @Fabrizio69 @maggia80

matejhof, 2022-12-02:
Thanks! This looks like something for @0xBEDA and @chatom98. We will try next week I guess.

AntonioConsilvio, 2022-12-05:
Hello @matejhof @rozliv, regarding torso skin, the problem should concern the 4 skin boards (MTB4) placed on torso cover (I circled cables that should have the problem). 
[image]

To check these problems you should remove torso cover, but these boards are placed like this:
[image]

So you should be able to check the status of the boards from the back and the bottom of the robot at this height, without removing covers:
[image]

If you have any other questions, write us. 

cc @Fabrizio69

matejhof, 2022-12-06:
Hi @AntonioConsilvio,
as for the right hand, we solved this problem now. We had a spare ring finger fingertip, so we replaced it and now it works - the jumper is back, joints seems fine, right hand skin is working. thanks to @0xBEDA and @chatom98!

matejhof, 2022-12-06:
As for the torso ski, what kind of diagnostics should we do? All the 4 boards are blinking in the same way - blue LED is on, red one is blinking. connectors look fine. @AntonioConsilvio ?

matejhof, 2022-12-06:
Reading from the port can tell us something - the values do not change - they are not live. 
[image]
If we read from other ports, there are small changes even if noone touches the skin.
On the compensated port, there is nothing going on at all:
[image]
Thanks, @AntonioConsilvio.

AntonioConsilvio, 2022-12-07:
So the torso skin problem persist.

> Then you have to do the same thing you did before with `FirmwareUpdater`, but after clicking on Discover, you should select the board mc4plus with address 10.0.1.22. After that, click on Force ETH [...]
But if boards are blinking, it means that are turn on. So if these boards are on, you should be able to see them on `FirmwareUpdater` by following these steps previously listed.

But when you did that, you don't see them, how we can see in this comment below:

> Hi @AntonioConsilvio, Here are the terminal output as txt file and the screenshot of FirmwareUpdater.
> 
> [output.log](https://github.com/robotology/icub-tech-support/files/10069278/output.log) [cleaned version](https://github.com/robotology/icub-tech-support/files/10069407/new.log)
> 
> [image]

So maybe retry to follow those steps and check if skin boards (MTB4) are visible on `FirmwareUpdater`. 

cc @Fabrizio69

rozliv, 2022-12-08:
Hi @AntonioConsilvio,
the boards are not visible on FirmwareUpdater:
`FirmwareUpdaterCore::getCanBoardsFromEth(): No CAN boards found beneath  "10.0.1.22"  after a successful driver init.`

Errors from yarplogger: 
`	1320.548923	ERROR			 from BOARD 10.0.1.22 (face-eb22-j0), src CAN1, adr 0, time 1478s 642m 123u: (code 0x00000019, par16 0x0901 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
`

@matejhof @0xBEDA

AntonioConsilvio, 2022-12-14:
Hi @rozliv @matejhof, I think  the problem is the cable **W_N9** (Orange), which connects mc4plus (EB22, Blue) to the first MTB4 (9B7, Green) of the torso: 

[image]

[image]

So, you should be able to check the connector on MTB4 (green) from the bottom of the torso: 

[image]

If the connector on MTB4 is fine, the problem should be the connector on mc4plus (blue), which should be the first of the three boards inside the head, as you can see in the photo below:

[image]

If you have to open the head to check the connector on mc4plus, to figure out where the cable is, there are colored cubes on the cables, check these cubes and find the cable with this name: **W_N9**.

Write to us if you have any questions.

cc @Fabrizio69

rozliv, 2022-12-17:
Hi @AntonioConsilvio, 
just checking the cable and connector on mc4plus made it work. Thank you!

@matejhof

closed 2022-12-17 by rozliv