robotology/mesh-support #1457: CPU button is broken and cannot be pressed

gabrielenava, 2022-11-10:
### Device name 🤖

iCubGenova04

### Request/Failure description

the CPU button on the backpack is broken and it can barely be pressed. In Milan we were pressing it with the help of an allen key.

### Detailed context

Is it possible to replace it, and verify that there are no other damages to the power board?

AntonioConsilvio, 2022-11-17:
Hello @gabrielenava, the CPU button has been replaced and now works without problems. I await feedback to close the issue.

cc @Fabrizio69 @maggia80

gabrielenava, 2022-11-18:
hi @AntonioConsilvio, the button works fine now! we can close the issue

closed 2022-11-18 by AntonioConsilvio