robotology/mesh-support #1458: R1 does not start w/ latest devel

marcoaccame, 2022-11-11:
### Device name 🤖

R1

### Request/Failure description

I open a ticket from what @randaz81 has told me. Pls @randaz81 complete or change the following info as you like.

On October 20, I receive from @randaz81 the following picture telling that yarprobotinterface does not start w/ latest devel.

[image]

After that, we did tests on our setups and we did not see the problem.

Nevertheless, on R1 `yarprobotinterface` still does not start. 

### Detailed context

pls @randaz attach a log.

### Additional context

The system uses `docker`.

randaz81, 2022-11-15:
About the boards' firmware:
[image]

This is the SHA of icub-main which is **working**:
[image]

This is the SHA of icub-firmware-shared which is **working**:
[image]

If I use the latest devel for icub-main/icub-firmware-shared instead, I get the cryptic error mentioned above.

Please provide support.

pattacini, 2022-11-15:
As agreed w/ @randaz81, we will start processing this request the next week as we're now busy w/ ergoCub SN000.

marcoaccame, 2022-11-15:
Thanks @randaz81. I will look at that as soon as possible.

pattacini, 2022-11-24:
Hi @randaz81 

We experienced a very similar problem while fiddling with the ergoCub binaries, which were not up-to-date and pretty much of the same age as yours.

However, by flashing the very latest devel versions of the binaries, we didn't spot the problem again.
Thus, while we're doing our investigation to narrow down the cause, you may consider flashing the latest devel.

marcoaccame, 2022-11-28:
Hi all, here are some tests on the latest distros.

@sgiraz and I have verified  `yarprobotinterface` in various distros starting from `2022.08.0` until the latest `2022.09.1`. 

We tested two different setups: 

- one with an `ems`  running the multiple FT service (laptop, `ems `board, `strain2 `board), 
- one with an `mc4plus` running the MC service (laptop, `mc4plus `board, DC motor).

They all behaved in the correct way. In here a summary of the results

| DISTRO    | RESULT | YARP   | icub-main | icub-fw-shared | icub-fw-build | ems   | mc4plus |
| --------- | ------ | ------ | --------- | -------------- | ------------- | ----- | ------- |
| 2022.09.1 | OK     | v3.7.2 | v1.28.1   | v1.27.0        | v1.27.0       | v3.54 | v3.52   |
| 2022.09.0 | OK     | v3.7.2 | v1.28.1   | v1.27.0        | v1.27.0       | v3.54 | v3.52   |
| 2022.08.1 | OK     | v3.7.2 | v1.28.1   | v1.27.0        | v1.27.0       | v3.54 | v3.52   |
| 2022.08.0 | OK     | v3.7.2 | v1.27.0   | v1.26.0        | v1.26.0       | v3.53 | v3.51   |

For more details see the page of the label required for each DISTRO: https://icub-tech-iit.github.io/documentation/sw_versioning_table/#software-versioning-table

sgiraz, 2022-11-28:
Hi @randaz81, 

To solve the problem reported, you may try to use even the latest devel versions or one of the distros reported above.
Please provide us with feedback.

cc @pattacini @marcoaccame

marcoaccame, 2022-11-28:
> Hi @randaz81,
> 
> To solve the problem reported, you may try to use even the latest devel versions or one of the distros reported above. Please provide us with feedback.
> 
> cc @pattacini @marcoaccame

And yes, the latest devel versions of `icub-main`, `icub-firmware-shared` and `icub-firmware-build` also work. We have testde them w/ `YARP` v3.7.2.

marcoaccame, 2022-11-29:
Today w/ @randaz81 and @sgiraz we updated to the latest devel and the robot starts correctly.

So, for now I will close the issue.

closed 2022-11-29 by marcoaccame