robotology/mesh-support #1459: iCubGenova11 - The right arm fingers do not complete the calibration

xEnVrE, 2022-11-14:
### Device name 🤖

iCubGenova11

### Request/Failure description

As per the title, the finger of the right arm do not complete the calibration procedure.

They close but they don't go back in the home configuration as they should. 

Please check out the picture below.

[image]

### Detailed context

As per the title, the finger of the right arm do not always complete the calibration procedure.

They close but they don't go back in the home configuration as they should. 

Please check out the picture below.

[image]

### How does it affect you?

We cannot perform grasping action with the right arm.

xEnVrE, 2022-11-15:
This evening we tried the robot after the maintenance activity. Sometimes, the right hand fingers still do not calibrate.

AntonioConsilvio, 2022-12-12:
I fixed this problem, I tried `yarprobotinterface` and it never failed.

cc @Fabrizio69

closed 2022-12-12 by AntonioConsilvio