robotology/mesh-support #1460: iCubGenova11 - Errors while running the yarprobotinterface

xEnVrE, 2022-11-14:
### Device name 🤖

iCubGenova11

### Request/Failure description

During the execution of the `yarprobotinterface` we noticed several errors in the output. We report the errors below. We also attach the `yarprobotinterface` log.

A part from the `viscousPos` parameter missing from many groups, the most important ones seems those regarding the  2FOC **quadrature encoder phase broken** for the torso and some other errors regarding the encoders, i.e. `CFG: EOtheEncodeerReader cannot be configured: verification fails.`

### Detailed context

### torso-eb5-j0_2

```
 1744,719096 <ERROR>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 29s 47m 387u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase\
 broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + . 
```

### right_arm-eb28-j8_11

```
 1729,101208 <ERROR>  from BOARD 10.0.1.28 (right_arm-eb28-j8_11), src LOCAL, adr 0, time 13s 521m 200u: (code 0x05000016, par16 0x4001 par64 0x00000000000f0005) -> CFG: EOtheEncodeerReader \
cannot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 a\
nd errorcodes in p64&0xffff0000 + .
```

### left_arm-eb24-j4_7

```
 1727,933088 <ERROR>  from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src LOCAL, adr 0, time 12s 351m 394u: (code 0x05000016, par16 0x4008 par64 0xfffffffff0005000) -> CFG: EOtheEncodeerReader ca\
nnot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and\
 errorcodes in p64&0xffff0000 + .
```

```
1727,882790 <ERROR> viscousPos parameter not found for board left_arm-eb24-j4_7 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlU\
nits machine_units) (kp -1000 0 0 0) (kd 0 0 0 0) (ki 0 0 0 0) (maxOutput 3360 0 0 0) (maxInt 500 0 0 0) (ko 0 0 0 0) (stictionUp 0 0 0 0) (stictionDown 0 0 0 0) (kff 0 0 0 0) (kbemf 0 0 0 0\
) (filterType 0 0 0 0) (ktau 1 0 0 0)
```

### right_arm-eb29-j12_15

```
 1774,057887 <ERROR> A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD right_arm-eb29-j12_15 IP 10.0.1.29, joint 0, current mode: cal, requested: i\
dl                                                                                                                                                                                             
 1774,057921 <ERROR> failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=idl) for BOARD right_arm-eb29-j12_15 IP 10.0.1.29 after 17 attempts and 0.262487 seconds     
 1774,057932 <ERROR> In embObjMotionControl::setControlModeRaw(j= 0 , mode= idl ) for  BOARD right_arm-eb29-j12_15 (IP 10.0.1.29)   has failed checkRemoteControlModeStatus() 
```

###  left_arm-eb1-j0_3

```
1727,651165 <ERROR> viscousPos parameter not found for board left_arm-eb1-j0_3 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlUn\
its machine_units) (kp 200 200 250 300) (kd 0 0 0 0) (ki 0 0 0 0) (maxOutput 8000 8000 8000 8000) (maxInt 500 500 500 500) (ko 0 0 0 0) (stictionUp 0.5 0.5 1 1.69999999999999995559) (stictio\
nDown -0.699999999999999955591 -0.699999999999999955591 -0.800000000000000044409 -1.19999999999999995559) (kff 1 1 1 1) (kbemf 0.00300000000000000006245 0.000599999999999999947438 0.00069999\
9999999999992888 0.000699999999999999992888) (filterType 0 0 0 0) (ktau 180 464 463 449) 
```

### left_arm-eb25-j8_11
```
 1728,155020 <ERROR>  from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 12s 577m 206u: (code 0x05000016, par16 0x4001 par64 0x00000000000f0005) -> CFG: EOtheEncodeerReader c\
annot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 an\
d errorcodes in p64&0xffff0000 + .
```

### right_arm-eb3-j0_3

``` 
1728,598488 <ERROR> viscousPos parameter not found for board right_arm-eb3-j0_3 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlU\
nits machine_units) (kp -50 -200 -250 -300) (kd 0 0 0 0) (ki 0 0 0 0) (maxOutput 8000 8000 8000 8000) (maxInt 500 500 500 500) (ko 0 0 0 0) (stictionUp -0.5 -0.5 -1 -1.69999999999999995559) \
(stictionDown 0.699999999999999955591 0.699999999999999955591 0.800000000000000044409 1.19999999999999995559) (kff 1 1 1 1) (kbemf -0.00300000000000000006245 -0.000599999999999999947438 -0.0\
00699999999999999992888 0.000699999999999999992888) (filterType 0 0 0 0) (ktau -180 -464 -463 -449) 
```

### torso-eb5-j0_2

```
 1729,545761 <ERROR> viscousPos parameter not found for board torso-eb5-j0_2 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlUnits\
 machine_units) (kp -450 -400 -400) (kd 0 0 0) (ki 0 0 0) (maxOutput 8000 8000 8000) (maxInt 500 500 500) (ko 0 0 0) (stictionUp 0 0 0) (stictionDown -2 -2 -2) (kff 1 1 1) (kbemf -0.00160000\
000000000007667 -0.00300000000000000006245 -0.00300000000000000006245) (filterType 0 0 0) (ktau -200 -200 -200)
```

### left_leg-eb6-j0_3 

```
1729,797900 <ERROR> viscousPos parameter not found for board left_leg-eb6-j0_3 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlUn\
its machine_units) (kp -200 200 0 -200) (kd 0 0 0 0) (ki 0 0 0 0) (maxOutput 8000 8000 8000 8000) (maxInt 500 500 500 500) (ko 0 0 0 0) (stictionUp -2.0 2.0 -1.0 -1.0) (stictionDown 1.399999\
99999999991118 -2.0 1.30000000000000004441 0.200000000000000011102) (kff 1 1 1 1) (kbemf 0 0 0 0) (filterType 0 0 0 0) (ktau -162 178 -197 -167)
```

### left_leg-eb7-j4_5

```
1730,030226 <ERROR> viscousPos parameter not found for board left_leg-eb7-j4_5 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlUn\
its machine_units) (kp 200 200) (kp 0 0) (kd 0 0) (ki 0 0) (maxOutput 8000 8000) (maxInt 500 500) (ko 0 0) (stictionUp 1.69999999999999995559 0.800000000000000044409) (stictionDown -1.699999\
99999999995559 -1.80000000000000004441) (kff 1 1) (kbemf 0 0) (filterType 0 0) (ktau 209 160)
```

### right_leg-eb8-j0_3

```
 1730,241914 <ERROR> viscousPos parameter not found for board right_leg-eb8-j0_3 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlU\
nits machine_units) (kp 200 -200 0 300) (kd 0 0 0 0) (ki 0 0 0 0) (maxOutput 8000 8000 8000 8000) (maxInt 500 500 500 500) (ko 0 0 0 0) (stictionUp 2.29999999999999982236 -1.7900000000000000\
3553 1.80000000000000004441 1.19999999999999995559) (stictionDown -3.29999999999999982236 1.76000000000000000888 -1.39999999999999991118 -1.69999999999999995559) (kff 1 1 1 1) (kbemf 0 0 0 0\
) (filterType 0 0 0 0) (ktau 147 -180 217 282)
```

### right_leg-eb9-j4_5 

``` 
1730,463914 <ERROR> viscousPos parameter not found for board right_leg-eb9-j4_5 in bottle TRQ_PID_DEFAULT (controlLaw torque) (outputType pwm) (fbkControlUnits metric_units) (outputControlU\
nits machine_units) (kp -200 -200) (kd 0 0) (ki 0 0) (maxOutput 8000 8000) (maxInt 500 500) (ko 0 0) (stictionUp -1.39999999999999991118 -1.5) (stictionDown 2.39999999999999991118 1.60000000\
000000008882) (kff 1 1) (kbemf 0 0) (filterType 0 0) (ktau -231 -180)  
```

### Additional context

Please find here attached the log from the `yarprobotinterface`.

[log_icub-head_yarprobotinterface_9427.txt](https://github.com/robotology/icub-tech-support/files/10007328/log_icub-head_yarprobotinterface_9427.txt)

pattacini, 2022-11-15:
I don't seem to have access to the iCubGenova11 conf files yet.

At any rate, I think the FW versions may be old as the problem with the missing `viscousPos` param was already addressed lowering the severity to a warning and relying on default values.

Regarding the quadrature encoder, we may need to clean it up.

@AntonioConsilvio @Fabrizio69 what kind of distro has been installed on iCubGenova11? Also, it'd be helpful to have access to the conf files, which in turn need to be hosted on our main robots-configuration repo via a PR.

cc @marcoaccame @sgiraz

xEnVrE, 2022-11-15:
Most of the errors have disappeared excluding those related to `quadrate encoder phase broken`. However, as @maggia80 told us this only depends on the firmware and it has been solved, if I got it correctly, in more recent releases of the software.

pattacini, 2022-11-16:
The most recent distro solved the `viscousPos` error, whereas the quadrature encoder error can be disregarded as it's not employed in the control for this robot.

I reckon we can close for now the issue.

xEnVrE, 2022-11-16:
Closing then.

closed 2022-11-16 by xEnVrE