robotology/mesh-support #1462: iCubGenova11 - Warped range of motion for both thumb opposition joints

xEnVrE, 2022-11-15:
### Device name 🤖

iCubGenova11

### Request/Failure description

The range of motion of both hands thumb opposition is warped in that when commanding the joint towards 90 degrees the joint moves way beyond 90 degrees (check out the picture)

[image]
> this is the left hand

We also noticed that while using the finger the range is still warped but it expands or decompresses. I.e., sometimes we can only move between say, ~12 degrees and 25 degrees. Some other times it expands to a different range between say, ~12 degrees and ~ 50 degrees.

For this reason, it seems that it is not just an offset in the calibration. Rather, the magnet of the hall effect sensor is not placed properly as the readings are not linear with respect to the joint motion.

### Detailed context

The range of motion of both hands thumb opposition is warped in that when commanding the joint towards 90 degrees the joint moves way beyond 90 degrees (check out the picture)

[image]
> this is the left hand

We also noticed that while using the finger the range is still warped but it expands or decompresses. I.e., sometimes we can only move between say, ~12 degrees and 25 degrees. Some other times it expands to a different range between say, ~12 degrees and ~ 50 degrees.

For this reason, it sees that it is not just an offset in the calibration. Rather, the magnet of the hall effect sensor is not placed properly as the readings are not linear with respect to the joint motion.

pattacini, 2022-11-18:
This should be fixed.
Could you confirm that @AntonioConsilvio?

AntonioConsilvio, 2022-11-18:
Yes @pattacini, I turned the magnet in both thumbs, and recalibrated. I tried several times and seems work without problems.

cc @Fabrizio69

pattacini, 2022-11-18:
Well done!
Closing then.

closed 2022-11-18 by pattacini