robotology/mesh-support #1463: iCubGenova11 - The torso roll goes in hardware fault

xEnVrE, 2022-11-15:
### Device name 🤖

iCubGenova11

### Request/Failure description

While moving the robot arms and torso during a grasping experiment, the robot torso roll often goes in hardware fault.

### Detailed context

We noticed the following message from the `yarprobotinterface`:

```
[ERROR]  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 5289s 993m 145u: (code 0x02000002, par16 0x0000 par64 0x0000000000000000) -> MC: i2t limit exceeded. The motor has been turned off to prevent it from being damaged by overheating due to a continuous high current. par16 = ID of joint. + .
```

### How does it affect you?

We cannot perform upper body motions involving the torso roll.

valegagge, 2022-11-15:
Tomorrow we'll check the I2T limits in the configuration files. Maybe we can increase them a bit.

maggia80, 2022-11-16:
@AntonioConsilvio @GiacomoVassallo13 you could check the tension of the tendons moved by the motors `3M0 `and `3M1`. If they are too tight the motor uses more current than needed.

AntonioConsilvio, 2022-12-13:
Hi @xEnVrE, I used the robot's torso several times and it doesn't give me problems anymore. 

However, I await that you will run your Demo to see if this problem was solved with some torso movement or not.

cc @Fabrizio69

xEnVrE, 2022-12-13:
Thanks @AntonioConsilvio 

I'll let you know how it goes, probably next week.

xEnVrE, 2023-01-19:
We made enough tests and we are happy with that. 

We can close the issue.

closed 2023-01-19 by maggia80