robotology/mesh-support #1464: Yarprobotinterface freezes after several minutes of work

gabrielenava, 2022-11-16:
### Device name 🤖

iCubGenova04

### Request/Failure description

I report this problem occurred during our experiment in the wind tunnel in order to document it in case it happens again on the robot. Unfortunately, it is not easy to reproduce.

### Detailed context

The wind tunnel experiments last two days, and we run the robot for several hours every day. There was a MATLAB controller moving the robot while in position direct. More than once, it happened that after several minutes of work the MATLAB controller failed to set the control mode back to position. When this happened, we noted that actually the `yarpmotorgui` and `yarpmanager` were freezed, and we cannot switch the robot off anymore from the laptop. icub-head was still reachable (we were able to ping it normally).

The only solution at that point was to turn off and on the motors (sometimes we also restarted icub-head, but not sure it was necessary) several times, until finally the robot restarted. In the failed startup attempts, we were getting error messages from all the boards (I saved a couple of loggers, I will upload them to the issue).

We don't know the origin of this problem, the only custom part of the setup was an extension of the fault button that the wind tunnel technicians made in order to be able to fault the robot from the control room. If it happens also in the lab, we can try to debug it.

### Additional context

I will upload the logger files I saved from the campaign

AntonioConsilvio, 2022-11-17:
Hello @gabrielenava, the error about the boards should be caused by the FT sensor (strain) inside the left foot. Because in the electrical cable **7N2_0** the white and green cables were detached. 

When I fixed this, the Firmwareupdater show me the strain and the robot has been tested and seems to be working. 

I await feedback to close the issue.

cc @Fabrizio69 @maggia80

gabrielenava, 2022-11-18:
we tested the robot in a demo on 17/11 and we did not get any issue. 

I think we can close the issue for now, at least the problem has been documented. If and when it will happen again, we will address it.