robotology/mesh-support #1465: The robot powers off while position controlled

gabrielenava, 2022-11-18:
### Device name 🤖

iCubGenova04

### Request/Failure description

today we were checking iCubGenova04 on board pc and running some others routine checks. Suddenly, the robot power shut down. Both CPU and MOTORS button were grey (no power). After a while, the power came back.

This happened more than once. We checked the power supply and the power cable: they are ok.

### Detailed context

the same problem happened once with @Fabrizio69 and @AntonioConsilvio two days ago while repairing the robot for #1457 

in that occasion, we were using a different power supply and a different power supply cable. We were debugging issues related to the fault system, and once the robot shuts down completely in a similar way.

gabrielenava, 2022-11-18:
cc @HosameldinMohamed

maggia80, 2022-11-19:
I think the reason could be the fake battery that is not mounted correctly and loses contact from time to time.

HosameldinMohamed, 2022-11-19:
> I think the reason could be the fake battery that is not mounted correctly and loses contact from time to time.

Yes, at the end of the day @AntonioConsilvio found that the fake battery was a bit loose and he tightened the scrwes. Now it seems OK!

Fabrizio69, 2022-11-21:
Thanks @HosameldinMohamed, we close this issue

cc: @AntonioConsilvio

closed 2022-11-21 by Fabrizio69