robotology/mesh-support #1466: iCubLisboa01 - right wrist flexion does not calibrate properly

plinioMoreno, 2022-11-21:
### Device name 🤖

iCubLisboa01

### Request/Failure description

During robot start up with `yarprobotinterface`, the right wrist flexion does not move properly to home position. The rest of the right hand joints seem to work, but it takes longer than expected.

Log file of yarprobotinteface in the following:

[log_icub-head_yarprobotinterface_4869.txt](https://github.com/robotology/icub-tech-support/files/10055874/log_icub-head_yarprobotinterface_4869.txt)

### Detailed context

The right wrist flexion does not start properly, it seems that there is some intermittent communication or commands to the motor, but it does not calibrate properly

AntonioConsilvio, 2022-11-29:
Hi @plinioMoreno, with this information I can think that there is a problem between the electrical wires connecting the mc4_plus board (EB27) to the endcoder (AEA 5 or 6, I think 6).

To understand better I need to know if when starting `yarprobotinterface` and then opening `yarpmotorgui`, if you manually move your right wrist, does the encoder value change or not?

With `yarpmotorgui` you should be able to tell exactly which joint has this problem. It should be either 4, 5 or 6, but I think it is 6. So check if encoder value changes and if it does this, by how much.  

cc @Fabrizio69

plinioMoreno, 2022-12-05:
Hi @AntonioConsilvio, we run `yarprobotinterface' and looked at the encoder values in `yarpmotorgui`. Your guess is correct, the encoder of joint 6 does not change its value at all, it keeps the same value all time. Let us know what is the next step.

AntonioConsilvio, 2022-12-07:
Hi @plinioMoreno you should open robot's covers. Because, as you can see in the picture below, AEA6 sensor is under palm (blue), then there is a cable (orange) that connects AEA sensor to mc4plus board (EB27, green). 
[image]

So you should open the palm and check the soldering on AEA sensor (send a photo). 

If the soldering is fine, you should open upper-arm cover to check the cable connector. There are 3 mc4plus boards under upper-arm cover, the board to which the cable is connected is EB27 (green) and to find the cable you need to look the cable called R2S2 (there are colored cubes on the cables, check these cubes and find the cable with this name)
[image]

But, before open upper-arm cover, send a photo of the soldering under the palm.

Are you able to do these things? 
If you have any question, write us.

cc @Fabrizio69

plinioMoreno, 2022-12-07:
Hi @AntonioConsilvio, thank you for the instructions. With the help of @Ryo-Naohara and @wevmacedo we found that the encoder cable was broken next to the board `EB27`. Now it is working properly.

closed 2022-12-07 by plinioMoreno