robotology/mesh-support #1467: iCubGenova09 - Right thumb not calibratring

S-Dafarra, 2022-11-22:
### Device name 🤖

iCubGenova09

### Request/Failure description

After it broke in LA, the right thumb was not calibrating properly

### Detailed context

The right thumb did not calibrate properly

GiacomoVassallo13, 2022-11-22:
Hi the thumb distal joint was replaced. Distal was blocked because was the teflon sheath of the cable that had peeled off.

cc: @AntonioConsilvio @S-Dafarra

AntonioConsilvio, 2022-11-23:
Me and @S-Dafarra tested all the finger joints and they seem to be working, including the distal joint of the right thumb. So everything seems fine.

@Fabrizio69 @GiacomoVassallo13