robotology/mesh-support #1470: Eyelids motor broken

S-Dafarra, 2022-11-29:
### Device name 🤖

iCubGenova09

### Request/Failure description

The eyelids do not move. The mechanism seems stuck

### Detailed context

It seems that the motor is broken. It gives a lot of resistance even without power. It probably broke after the flight to Los Angeles.

GiacomoVassallo13, 2022-12-13:
Hi. The eyelid motor gearbox appeared slightly unscrewed, so it can be deduced that the coupling was not well aligned This could have caused damage to the gearbox. 
The eyelid motor has been replaced and tested.

cc: @Fabrizio69 @AntonioConsilvio

S-Dafarra, 2022-12-13:
Unfortunately, the eyelids still do not move, or they are extremely slow.

AntonioConsilvio, 2022-12-13:
> Unfortunately, the eyelids still do not move, or they are extremely slow.

We will try recalibrating and running the eyelids without face cover to see if any of these elements are causing problems or if the problem is the motor.

cc @Fabrizio69 @S-Dafarra @GiacomoVassallo13

AntonioConsilvio, 2022-12-20:
I @S-Dafarra, we recalibrated and tested the motor with @GiulioRomualdi.

It works well and without problems.

cc @Fabrizio69 @GiacomoVassallo13

closed 2022-12-20 by AntonioConsilvio

GiacomoVassallo13, 2022-12-20:
Hi, The movement of the eyelids was tested in control with the robot without face cover. The slow motion problem was due to the fact that the motor could not reach a certain value in the position due to so much effort and overheating. By repositioning the motor, it was able to reach the position, thus making the motor able to move correctly and no longer overheats.

cc: @Fabrizio69 @AntonioConsilvio