robotology/mesh-support #1473: Right wrist tendons broken

S-Dafarra, 2022-11-30:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right wrist tendons appear to be broken. I am not sure how this could have happened.

### Detailed context

https://user-images.githubusercontent.com/18591940/204843627-732180a6-9e32-456a-9250-4cef6592a02e.mp4

GiacomoVassallo13, 2022-12-01:
Hi, the passive tendons of wrist has been replaced.

cc: @AntonioConsilvio @S-Dafarra

S-Dafarra, 2022-12-02:
Hi @GiacomoVassallo13, I think there still an issue. The tendon seems out of place

[image]

S-Dafarra, 2022-12-07:
The wrist seems to be fully functional again!