robotology/mesh-support #1474: Left hand - fingers abduction does not move

plinioMoreno, 2022-12-05:
### Device name 🤖

iCubLisboa01

### Request/Failure description

The left hand fingers abduction joint (`l_hand_finger`) does not move after starting the robot. If we try to move the motor on `yarpmotorgui` the motor does not move, stays at the same location. 

### Detailed context

The following video shows the `yarpmotorgui` screen:

https://user-images.githubusercontent.com/3798890/205710153-a75cd754-1b7c-4ef4-83ba-43e6668dfd70.mp4

The log from `yarprobotinterface` as follows:

[log_icub-head_yarprobotinterface_3000.txt](https://github.com/robotology/icub-tech-support/files/10156741/log_icub-head_yarprobotinterface_3000.txt)

Please ignore the errors from the right arm, which are part of other issue.

### How does it affect you?

We cannot run grasping experiments with the hands.

plinioMoreno, 2022-12-07:
Hi @AntonioConsilvio,

We removed the cover of the left hand to see if there was something wrong with this joint. After placing the hand cover back, now the joint (`l_hand_finger`) goes to fault during startup. The log from `yarprobotinterface` as follows:

 
[log_icub-head_yarprobotinterface_3279.txt](https://github.com/robotology/icub-tech-support/files/10177699/log_icub-head_yarprobotinterface_3279.txt)

Let us know how to fix this problem.

cc @Ryo-Naohara

AntonioConsilvio, 2022-12-12:
Hi @plinioMoreno, ok this shouldn't be an electrical problem. 
So, try to remove the back of the hand and with a slotted screwdriver try to manually move the motor (green, in the photo below).
[image]

If, when you move the motor with the screwdriver, fingers don't move it means there is a problem with the steel cable (whose condition you can check by following the blue arrow in the picture above) or with the motor.

cc @Fabrizio69

plinioMoreno, 2022-12-19:
Hi @AntonioConsilvio, this is the video that shows that the fingers move correctly.

https://user-images.githubusercontent.com/3798890/208464702-5b1db110-2e32-483d-bb54-f81eb6d49d77.mp4

It seems that the cable is fine. However, we still have the behavior described [here](https://github.com/robotology/icub-tech-support/issues/1474#issue-1477133521)

AntonioConsilvio, 2023-01-02:
Hi @plinioMoreno, I am not sure about the cause of the problem, but you can try checking the connection of the Hall sensor (located above the little finger, blue), whose cable (orange) go to the EB24 board (green).

[image]

(The photo above is only for visual inspection, do not touch these cables if they appear as in the photo above)

[image]

To find the cable, you need to check the colored cubes on the cable and find the cable called 2M3. The EB24 board is the first of the three board present in the left arm.

[image]

Write to us if you have any questions. 

cc @Fabrizio69

AntonioConsilvio, 2023-01-20:
Hi @plinioMoreno, I propose that you schedule a videocall on Monday, February 23 at 10:00 a.m. in your time zone.

Are you available on this date?

cc @Fabrizio69

plinioMoreno, 2023-01-20:
Hi @AntonioConsilvio, that sounds good. I confirm that @Ryo-Naohara is available to talk with you on the coming monday. Do we need to setup something on our icub laptop?

AntonioConsilvio, 2023-01-20:
> Do we need to setup something on our icub laptop?
@plinioMoreno This should not be necessary, just have the laptop ready to start the robot. I suggest keeping the back of the robot's hand and the left upper-arm cover open.

AntonioConsilvio, 2023-01-26:
Hi @plinioMoreno, after the video call we suspect that the motor needs to be replaced. 
So, analyzing the situation, and the other numerous problems found on the left hand, we wanted to suggest that you send us the robot arm so that we can fix all the problems present on the hand. 

If you agree, I will show you how to separate the arm from the robot.

cc @Fabrizio69

plinioMoreno, 2023-01-26:
Hi @AntonioConsilvio,

Do we have to remove the full arm? Or just the forearm? I am asking this because we also found some missing tactile information on the right hand.

AntonioConsilvio, 2023-01-27:
@plinioMoreno I'm sorry, I didn't quite understand. You should send us almost the whole arm. You should remove those 4 screws (red) and remove some connectors from the electronics boards. I can tell you which connectors if you want to send us the left arm.

[image]

However, I did not understand why you are talking about the right hand, do you have problems with the right hand as well? If you have problems with it, what problems do you have?

Finally, I did not understand whether you agree to send us the left arm or not. 

Basically, I would need more precise information. Thank you.

cc @Fabrizio69

plinioMoreno, 2023-01-27:
Hi @AntonioConsilvio,

Sorry for the multiple information. Regarding the right hand, yes, we are having problems with the taxels of the thumb, where the values do not change in the skinGui. Ryo will open an issue to see if we solve that problem with your help.

Regarding the left hand/arm, we agree with sending you the arm during next week. We will remove the arm and send it to you.

AntonioConsilvio, 2023-01-30:
Hi @plinioMoreno, before activating shipping, perhaps it may be helpful to know the right arm problem, to see if it may be smart to send us both front arms, or if the right arm problem can be solved with remote support.

cc @Fabrizio69

plinioMoreno, 2023-01-30:
Hi @AntonioConsilvio,

The right arm problem is described in [this issue](https://github.com/robotology/icub-tech-support/issues/1493#issue-1562550875).

plinioMoreno, 2023-02-15:
Hi @AntonioConsilvio, after the final diagnostic of the right hand [here](https://github.com/robotology/icub-tech-support/issues/1493#issue-1562550875), we are going to send just the left arm. @Ryo-Naohara unmounted the arm and we are ready to send it to you. Do you want to send the address and sender info please?

andreaderito, 2023-02-15:
Hi @plinioMoreno the address where to send the parcel is:
IIT - Istituto Italiano di Tecnologia
Via Morego 30
I-16163 Genoa, ITALY
Att.n: Fabrizio Larosa (iCub Tech, CIRS)
Phone: +39 010 28 98 223

**IMPORTANT: please safely pack the goods using the maximum care to absorb shocks**. If possible, indicate "**fragile**" on the parcel. As soon as you have a **tracking number**, pls let us know it so we may follow the shipping. **If possible, ensure the goods with the shipping company** - indicative (reasonable) values you may put 30keur for 1 lower-arm (forearm, wrist and hand altogether).
If you need further info, pls let me know.

cc @Fabrizio69 @AntonioConsilvio @maggia80

andreaderito, 2023-03-07:
Ok, thank you. Just in case you have a tracking number, pls let me have it and if it is DHL or TNT or other - if you don't have it handy, no problem of course - thx and best.

plinioMoreno, 2023-03-07:
I do not have tracking number info, but this is the picture:
[image]

andreaderito, 2023-03-09:
Hi Plinio, we received it - in the next day (or days, as soon as my colleagues are available), we'll open it and start working on it. Thanks!
cc @Fabrizio69 @maggia80 @AntonioConsilvio

plinioMoreno, 2023-03-10:
Hi @andreaderito,

Good that arrived safely, and thanks for letting us know!

AntonioConsilvio, 2023-03-15:
Hi @plinioMoreno, we opened the package and we noticed that the directions given in this [comment](https://github.com/robotology/icub-tech-support/issues/1474#issuecomment-1406230197) were not followed. 

It means that we haven't the electronics boards and that the pronosupination mechanism has been split in half.

It create two problems:
1. After the interventions, we cannot fully test the whole assembly, but only partially because we haven't the electronics boards.
2.  The way the pronosupination mechanism was disassembled was not according to the instructions provided, so the harmonic drive was damaged, and it needs to be replaced.

We show an example of a our ForeArm:
[image]

Your ForeArm shows like this:
[image]

So We need this mechanical part (RED in the photo below) that you should have. Because we have to change the harmonic drive and riassemble the Pronosupination mechanism.

[image]

Since you will send this mechanical part, you should also send the electrical boards, the following the instructions in this [comment](https://github.com/robotology/icub-tech-support/issues/1474#issuecomment-1406230197).

 If the instructions are not clear or if you have any question, please write us.

cc @sgiraz @Fabrizio69 @maggia80

plinioMoreno, 2023-03-15:
Hi @AntonioConsilvio,

First, we are sorry for the misunderstanding. By mistake @Ryo-Naohara thought that the screws to be removed are the ones marked with black circles:
[image]

We want to double check with you that the missing parts are the ones in the picture above (control boards EB24,EB25 and EB26). Then, we will let you know when the package is sent.

Also we want to understand why the harmonic drive got damaged, since all the parts seemed to be okay when removed.

AntonioConsilvio, 2023-03-15:
@plinioMoreno, Yes, the electronics boards are the ones in your photo.

Regarding the harmonic transmission, the Pronosupination mechanism has a specific sequence of assembly. So, consequently, it has a specific sequence of disassembly and in that sequence that mechanical part is the last one that has to be removed.

So, probably the harmonic transmission is probably damaged (we hope not) and to verify that we need that part, because even if it is not visibly damaged, we need to verify it to be sure.

cc @Fabrizio69 @sgiraz

Ryo-Naohara, 2023-03-15:
@AntonioConsilvio, is this the steel plate?
[image]

AntonioConsilvio, 2023-03-16:
@Ryo-Naohara, Yes it is the steel plate

plinioMoreno, 2023-03-23:
Hi @andreaderito and @AntonioConsilvio,

The missing parts of the left arm, control boards and steel plate were shipped yesterday through DHL. @andreaderito should have received an email, and the small box should arrive today to your facilities.

andreaderito, 2023-03-23:
Hi @plinioMoreno thanks, we'll inform you when the parcel arrives (no email up to now, but  absolutely no problem, we know the parcel is about to arrive so we just wait for it - thx).
cc @AntonioConsilvio

GiacomoVassallo13, 2023-03-23:
Good morning. the abduction joint motor was replaced and the tendon rewired. Once reconnected it will be electrically tested. 

[image]

cc: @sgiraz @Fabrizio69 @AntonioConsilvio

AntonioConsilvio, 2023-03-27:
Hi @andreaderito @plinioMoreno, the parcel has arrived and, at first glance, everything looks fine.

cc @sgiraz @Fabrizio69

sgiraz, 2023-05-11:
Hi @plinioMoreno,

## ⚙️ Maintenance 
We finalized all the activities on the left hand of `iCubLisbona01`. Check out the video below to see it in action:

| Startup  | Adduction |
| ------------- | ------------- |
| <video src="https://user-images.githubusercontent.com/114915464/235850884-13545cf3-4ba0-4a7c-9148-2034d5b60f23.mp4">  | <video src="https://user-images.githubusercontent.com/114915464/235850890-968eb69e-4196-48e4-8f56-0e0d6caeae3b.mp4"> |

| Skin test |
| ------------- |
| <video src="https://user-images.githubusercontent.com/114915464/235710142-c35694b0-82f8-44ab-885b-b174f2b8878a.mp4">  |

--- 

## 📦 Shipment 
As soon as it is shipped, you will receive the following package:
 
| Left hand | Fingertips + Spare wire | Full package |
| ------------- | ------------- | ------------- | 
| [image] | [image] | [image] |

> **Note**
> notes here

sgiraz, 2023-05-12:
I leave this issue in `review`.

@plinioMoreno let us know when you received the package.

andreaderito, 2023-05-14:
Hi @plinioMoreno I've acted so as to ask for the Documento di Trasporto (DDT, valid only in Italy) and also prepared the Packing List with the value of the goods (this file is already finalised, so pls find it here since now in case you wish starting preparing the insurance coverage of the goods). **The goods will be ready for pick up since next Tuesday, May 16th, 2023 at 08.30 from the address of IIT Central Warehouse i.e. IIT - Via Morego 30, 16163 Genoa, Italy**. Pls let us know when the goods arrive at your premises so we relax a bit - best, Andrea

[PACKING LIST (DETAILED_LIST_OF_GOODS)_v1_DEF.pdf](https://github.com/robotology/icub-tech-support/files/11472482/PACKING.LIST.DETAILED_LIST_OF_GOODS._v1_DEF.pdf)

cc @sgiraz @AntonioConsilvio @Fabrizio69 @maggia80

plinioMoreno, 2023-05-15:
@sgiraz ,thank you for the pictures and the videos. From them I see that everything looks good, thanks!

@andreaderito, if we want to do the insurance, how shall we proceed?

andreaderito, 2023-05-15:
Hi @plinioMoreno  normally you may select this option when you book the express courier (DHL or similar, pls check with your admin if they know how to do it - otherwise pls let me know, best

andreaderito, 2023-05-16:
Hi @Plinio, a small but important errata corrige: you may book the pick up of the goods starting from Thursday 18th morning (not from today, sorry). The pick up address is IIT Central Warehouse, Via Morego 30, 16163 Genoa, Italy (no change in respect to what indicated before). Best.
cc @Miche19

AntonioConsilvio, 2023-05-16:
Hi @plinioMoreno, remember to update the robots-configuration files, because I update the calibration of some joints after the maintenance service.

- https://github.com/robotology/robots-configuration/pull/501

cc @sgiraz

plinioMoreno, 2023-06-01:
We put the left arm back and everything is working as expected. Thanks @AntonioConsilvio for the reminder on the configuration files. From our side, this issue can be closed.

Also thanks to all the people involved in the repair.