robotology/mesh-support #1477: Left camera not working

S-Dafarra, 2022-12-12:
### Device name 🤖

iCubGenova09

### Request/Failure description

The left camera does not stream any image

### Detailed context

Here the output when launched from terminal
```
icub@icub-head-cam:~$ yarpdev --from camera/pylonCamera_config_left.ini --period 0.1 --width 800 --height 600
[DEBUG] |yarp.dev.PolyDriver|pylonCamera| Parameters are (device frameGrabber_nws_yarp) (from "camera/pylonCamera_config_left.ini") (height 600) (name "/icub/cam/left") (period 0.100000000000000005551115123125782702) (rotation 90.0) (rotation_with_crop false) (serial_number 40140941) (single_threaded 1) (subdevice pylonCamera) (width 800)
[WARNING] |yarp.device.frameGrabber_nws_yarp| 'capabilities' parameter not found or misspelled, the option available are COLOR(default) and RAW, using default
[INFO] |yarp.os.Port|/icub/cam/left/rpc| Port /icub/cam/left/rpc active at tcp://10.0.0.3:10179/
[INFO] |yarp.os.Port|/icub/cam/left| Port /icub/cam/left active at tcp://10.0.0.3:10184/
[DEBUG] |yarp.dev.PolyDriver|pylonCamera| Parameters are (device pylonCamera) (from "camera/pylonCamera_config_left.ini") (height 600) (name "/icub/cam/left") (period 0.100000000000000005551115123125782702) (pixelType 6449010) (rotation 90.0) (rotation_with_crop false) (serial_number 40140941) (single_threaded 1) (subdevice pylonCamera) (width 800)
[DEBUG] |yarp.device.pylonCamera| SERIAL NUMBER! 40140941 
[DEBUG] |yarp.device.pylonCamera| Setting  AcquisitionFrameRateEnable to true
[DEBUG] |yarp.device.pylonCamera| Setting  BslScalingEnable to true
[DEBUG] |yarp.device.pylonCamera| Setting  Width to 800
[DEBUG] |yarp.device.pylonCamera| Setting  Height to 600
[DEBUG] |yarp.device.pylonCamera| Setting  ExposureAuto to Off
[DEBUG] |yarp.device.pylonCamera| Setting  AcquisitionFrameRate to 10
[DEBUG] |yarp.device.pylonCamera| Starting with this fps 10
[DEBUG] |yarp.device.pylonCamera| Not using CUDA!
[INFO] |yarp.dev.PolyDriver|pylonCamera| Created device <pylonCamera>. See C++ class pylonCameraDriver for documentation.
[INFO] |yarp.dev.PolyDriver|pylonCamera| Created wrapper <frameGrabber_nws_yarp>. See C++ class FrameGrabber_nws_yarp for documentation.
[DEBUG] |yarp.dev.Drivers|pylonCamera| ===============================================================
[DEBUG] |yarp.dev.Drivers|pylonCamera| == Options checked by device:
[DEBUG] |yarp.dev.Drivers|pylonCamera| ==
[DEBUG] |yarp.dev.Drivers|pylonCamera| device=pylonCamera
[DEBUG] |yarp.dev.Drivers|pylonCamera| id [pylonCamera]
[DEBUG] |yarp.dev.Drivers|pylonCamera|     Id assigned to this device
[DEBUG] |yarp.dev.Drivers|pylonCamera| frameGrabber_nws_yarp.period=0.100000000000000005551115123125782702
[DEBUG] |yarp.dev.Drivers|pylonCamera|     refresh period(in s) of the broadcasted values through yarp ports
[DEBUG] |yarp.dev.Drivers|pylonCamera| frameGrabber_nws_yarp.capabilities
[DEBUG] |yarp.dev.Drivers|pylonCamera|     two capabilities supported, COLOR and RAW respectively for rgb and raw streaming
[DEBUG] |yarp.dev.Drivers|pylonCamera| frameGrabber_nws_yarp.no_drop
[DEBUG] |yarp.dev.Drivers|pylonCamera|     if present, use strict policy for sending data
[DEBUG] |yarp.dev.Drivers|pylonCamera| frameGrabber_nws_yarp.name=/icub/cam/left [/grabber]
[DEBUG] |yarp.dev.Drivers|pylonCamera|     name of port to send data on
[DEBUG] |yarp.dev.Drivers|pylonCamera| frameGrabber_nws_yarp.subdevice=pylonCamera
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.device=pylonCamera
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.id [pylonCamera]
[DEBUG] |yarp.dev.Drivers|pylonCamera|     Id assigned to this device
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.serial_number=40140941
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.width=800
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.height=600
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.period=0.100000000000000005551115123125782702
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.rotation=90.0
[DEBUG] |yarp.dev.Drivers|pylonCamera| subdevice.rotation_with_crop=false
[DEBUG] |yarp.dev.Drivers|pylonCamera| ==
[DEBUG] |yarp.dev.Drivers|pylonCamera| ===============================================================
[INFO] |yarp.os.Port|/icub/cam/left/quit| Port /icub/cam/left/quit active at tcp://10.0.0.3:10232/
[INFO] |yarp.dev.Drivers|pylonCamera| device active in background...
[INFO] |yarp.dev.Drivers|pylonCamera| device active in background...
[INFO] |yarp.dev.Drivers|pylonCamera| device active in background...
[INFO] |yarp.dev.Drivers|pylonCamera| device active in background...
[INFO] |yarp.dev.Drivers|pylonCamera| device active in background...
[INFO] |yarp.dev.Drivers|pylonCamera| device active in background...
[ERROR] |yarp.device.pylonCamera| Acquisition failed
*** Trying to allocate an invalid pixel type image
```

cc @pattacini @DanielePucci 

### How does it affect you?

Unfortunately, this is a bit critical since tomorrow we have a demo with the IIT Technical Committee

S-Dafarra, 2022-12-12:
> Unfortunately, this is a bit critical since tomorrow we have a demo with the IIT Technical Committee

We managed to use the only the right, so this is not too blocking for now

pattacini, 2022-12-13:
>*** Trying to allocate an invalid pixel type image

Somehow related: https://github.com/robotology/yarp/issues/2349.

cc @Nicogene

Nicogene, 2022-12-13:
It seems that the return of `getImage` is not checked:

https://github.com/robotology/yarp/blob/master/src/devices/frameGrabber_nws_yarp/FrameGrabber_nws_yarp.cpp#L265-L274

cc @randaz81 @elandini84

S-Dafarra, 2022-12-13:
> It seems an error when trying to acquire the images (the flat is damaged ?)

It is possible. The head covers had to be unmounted for example. Plus, when starting the device, the initial messages

```
[DEBUG] |yarp.device.pylonCamera| SERIAL NUMBER! 40140941 
[DEBUG] |yarp.device.pylonCamera| Setting  AcquisitionFrameRateEnable to true
[DEBUG] |yarp.device.pylonCamera| Setting  BslScalingEnable to true
[DEBUG] |yarp.device.pylonCamera| Setting  Width to 800
[DEBUG] |yarp.device.pylonCamera| Setting  Height to 600
[DEBUG] |yarp.device.pylonCamera| Setting  ExposureAuto to Off
[DEBUG] |yarp.device.pylonCamera| Setting  AcquisitionFrameRate to 10
[DEBUG] |yarp.device.pylonCamera| Starting with this fps 10
```
appear slower than the other one (that is working). Interestingly, it is the second time we have had an issue with the left, see https://github.com/icub-tech-iit/task-force-miscellanea/issues/138. Back in time, we blamed the movements of the bulb, but since then we stopped using those motors completely (not even during calibration).

S-Dafarra, 2022-12-13:
> I'm not in office an I cannot help you today. Are you using master? About the return value, if that is the problem and you have a urgent demo, I suggest to revert to a working commit.

I think that the issue is more on the electronics, which in turns causes the code to crash

randaz81, 2022-12-14:
> It seems that the return of `getImage` is not checked:
> 
> https://github.com/robotology/yarp/blob/master/src/devices/frameGrabber_nws_yarp/FrameGrabber_nws_yarp.cpp#L265-L274
> 
> cc @randaz81 @elandini84

Now fixed [b46623ba7b4c1e96e861adb8751ba049949c03cc](https://github.com/robotology/yarp/commit/b46623ba7b4c1e96e861adb8751ba049949c03cc)

S-Dafarra, 2022-12-14:
I would keep this open as the issue on the hardware is still there

maggia80, 2022-12-16:
@AntonioConsilvio on Monday, we should open the head and check the flat cables of the left camera. 

1) see if the led on the camera is on
2) check if the flat cables are well fitted in the connectors.

AntonioConsilvio, 2022-12-21:
Hi @S-Dafarra, I fixed the problem. A flat of left camera was quite detatched. After fixed this, the left camera started working but right camera stopped working. 

Then I changed the electronic board that connects the camera to CPU, after which both cameras start working.

[image]

cc @GiulioRomualdi @maggia80 @Fabrizio69 @pattacini

closed 2022-12-21 by AntonioConsilvio