robotology/mesh-support #1480: Ring finger loose due to gap

plinioMoreno, 2022-12-19:
### Device name 🤖

iCubLisboa01

### Request/Failure description

Close to the link between the ring finger and the palm, there is a gap between two parts. Video that illustrates the gap:

https://user-images.githubusercontent.com/3798890/208465998-79564010-1054-4d91-9283-fd86b232610b.mp4

### Detailed context

The ring finger does not seem to be aligned to the other fingers of the hand, due to a gap.

plinioMoreno, 2023-01-19:
Thanks @AntonioConsilvio, tightening the screw did not work, it seems that the part was damaged (bent), so I believe the solution is to bend back the section where the screw is. We will try that and let you know.

cc @Ryo-Naohara

sgiraz, 2023-05-12:
Well done @GiacomoVassallo13!

Maintenance activity completed, see:
- https://github.com/robotology/icub-tech-support/issues/1474#issuecomment-1543936332

Closing.

closed 2023-05-12 by sgiraz