robotology/mesh-support #1482: iCubGenova11 - Left thumb distal deviates when the opposition goes towards 90 degrees

xEnVrE, 2022-12-23:
### Device name 🤖

iCubGenova11

### Request/Failure description

As per the title. A picture of the problem follows:

[image]

### Detailed context

As per the title. A picture of the problem follows:

[image]

### How does it affect you?

It is not possible to command the thumb opposition and distal joints independently.

xEnVrE, 2022-12-23:
cc @fedeceola

AntonioConsilvio, 2022-12-23:
Hi @xEnVrE @fedeceola, this problem was caused by the sheath of the steel cable that was outside. Now the thumb can reach 90 degrees without that distal being pulled. 

Also, I recalibrate thumb oppose of left arm, so now both thumb oppose are in the same position when are in the same degrees.

If these problems occur again, you can reopen this issue.

cc @maggia80 @Fabrizio69

closed 2022-12-23 by AntonioConsilvio

xEnVrE, 2022-12-23:
Thanks @AntonioConsilvio