robotology/mesh-support #1483: Right arm shoulder tendon broke

rozliv, 2023-01-09:
### Device name 🤖

iCubPrague01

### Request/Failure description

The right arm shoulder tendon broke. @matejhof @0xBEDA

### Detailed context

While testing new arm movements, the right arm shoulder tendon broke.
[image]

matejhof, 2023-01-10:
hi @GiacomoVassallo13, we are on a tight schedule for a deadline and we need to run experiments. Our iCub purchase contract includes a maintenance plan for 3 years after warranty, which includes consumables (tendons, wires, crimps...). Could you please send us the tendon (or even better some extra ones) as soon as possible? Then we can maybe  organize a video call and you would explain to us how to change it? Thank you!

Fabrizio69, 2023-01-11:
Hi @matejhof, we are working to send you a spare shoulder set.

matejhof, 2023-01-12:
Thanks! 
While the set is on the way, can you please point us to instructions how to change it? Is there a tutorial? Or is it better to schedule a call with you and do it online under your supervision?   @0xBEDA will be in charge on our side.
@Fabrizio69 @GiacomoVassallo13

pattacini, 2023-01-12:
Hi @matejhof 

>While the set is on the way, can you please point us to instructions how to change it? Is there a tutorial?

In the meantime, you can peruse some resources at https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/.

rozliv, 2023-01-18:
We received tendons today. Now we are trying to replace them. We are stuck at replacing cables 1 and 2 together. We removed/loosed cables 5-8. The tendons seem to be too short when pulling them on the pulleys. While turning pulleys, the missing length varies from 3 to 20 mm. @Fabrizio69 @AntonioConsilvio @GiacomoVassallo13

AntonioConsilvio, 2023-01-18:
Hi @rozliv @matejhof , are you sure you put the right cable in the right place, following the right path around the pulleys?

Also, you should loosen the cable turnbuckles if you haven't already done so and keep the pulleys in the same position that you can see in the guide (https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/#41-right-shoulder-rotation-cable-1-and-2).  

It is difficult to understand and explain by message, so if you need more support you can definitely write to us; but if the problem cannot be solved, we can schedule a video call to provide better support.

Anyway keep writing to us if you need more suggestions.

cc @Fabrizio69 @GiacomoVassallo13

matejhof, 2023-01-18:
Hi! We suggest to schedule the video call - that seems like the most efficient solution. When would you be available? Thanks! @AntonioConsilvio @GiacomoVassallo13
 cc @Fabrizio69 @0xBEDA @rozliv

rozliv, 2023-01-18:
Here is a photo - the cable 2 is too short. [image]

AntonioConsilvio, 2023-01-18:
@rozliv Maybe try to rotate the pulleys so that they are exactly as in the picture.
[image]

AntonioConsilvio, 2023-01-18:
@rozliv To rotate this pulley, remove these two screws.
[image]

And if you have a pliers like this:
[image]

You can insert it into the two holes and rotate the pulley.

rozliv, 2023-01-18:
Those screws are not the original ones, we use them to rotate the pulley.

AntonioConsilvio, 2023-01-18:
Are the cable tensioners as loose as possible? @rozliv 
[image]

rozliv, 2023-01-18:
Do tensioners for cables 3 and 4 need to be loosened as well?

AntonioConsilvio, 2023-01-18:
Perhaps it is too difficult to explain by message, so I propose to schedule a video call on Monday, Feb. 23, at 10 a.m. to clarify the situation (if you are available).
@matejhof @rozliv 

cc @Fabrizio69 @GiacomoVassallo13

AntonioConsilvio, 2023-01-18:
@rozliv Yes, Tuesday we are available, I propose 10 a.m. if that's okay with you. 

Anyway you can try loosening cables 3 and 4, but I think in this situation it shouldn't help. Also because if it would help, it should be written in the guide.

cc @Fabrizio69 @GiacomoVassallo13

AntonioConsilvio, 2023-01-18:
> Tuesday at 11 is better for us. @AntonioConsilvio

Okay, that's fine for us too.

rozliv, 2023-01-24:
We tried to run the robot but we cannot move the shoulder forward (see the photo)
[image]

@Fabrizio69 @AntonioConsilvio

AntonioConsilvio, 2023-01-24:
@rozliv, the joint that doesn't work is the joint 0. The joint 0 is a motor that moves all pulleys. because:

[image]

As you can see in the photo above,  joint 0 hasn't a cable, there is a pin that moves other joints and therefore all the cables in the shoulder accordingly. 
So, if it cannot move, it means that (if the problem is mechanical) there is one of the other pulleys, where the cables turn, that is blocking the motor of joint 0.

So provide us with many pictures, from multiple viewpoints, of all the cable windings, especially this pulley:

[image]

But we need to know the situation of all the windings and any information that may be useful.

Another thing, when you start the robot, does the logger give errors?

cc @Fabrizio69 @GiacomoVassallo13

rozliv, 2023-01-24:
We loosed cables 1,2,7,8, then rotated the pulleys to be in the same position as on the other arm, and then tensioned the cables. Now it works. @AntonioConsilvio

matejhof, 2023-01-27:
Hello,
yesterday evening, the right shoulder tendons unfortunately broke again.  Now we know better how to fix it, but we're missing the tendons - especially number 1 that was missing in the original package (we used no. 3 instead, which has a different length). I gather that it was agreed during the call that you will send the no. 1 to us - is it already on the way? Please send two pieces if possible so we have spare. Thanks!   @AntonioConsilvio @Fabrizio69

Fabrizio69, 2023-01-27:
Hi @matejhof,
We are working on it, Andrea will send you the spare parts (two pieces for tendon n.1 and two pieces for tendon n.3

cc: @AntonioConsilvio

matejhof, 2023-02-06:
Thanks! Cables have arrived and the robot is fixed again. Closing!

closed 2023-02-06 by matejhof