robotology/mesh-support #1484: Right arm shoulder tendon broken

gabrielenava, 2023-01-12:
### Device name 🤖

iCubGenova04

### Request/Failure description

The right arm shoulder tendon broke while the robot was moving to the home position on the pole.

### Detailed context

cc @DanielePucci @S-Dafarra 

### How does it affect you?

we have a demo on Monday, would it be possible to repair the robot today or tomorrow morning?

AntonioConsilvio, 2023-01-13:
Hi @gabrielenava, @GiacomoVassallo13 and I replaced damaged cables (7 and 8) with two new ones. 

I await feedback on the operation of the robot.

cc @Fabrizio69

gabrielenava, 2023-01-13:
hello @AntonioConsilvio 

we tested the robot by running the Yoga++ demo several times. it worked with no issues!

thanks again, we can close the issue

closed 2023-01-13 by AntonioConsilvio