robotology/mesh-support #1485: Magnetometer of the IMU in the strain2 constantly streams zero

GiulioRomualdi, 2023-01-18:
### Device name 🤖

iCubGenova09

### Request/Failure description

Reading the magnetometer from IMU mounted on the strain2 through the yarp port I got a constant vector of zero elements. 

### Detailed context

I'm trying  to read the Magnetometer measurement from the IMU mounted on the FTs sensor of the  `iCubGenova09`. 
The `multipleanalogserver` device is used to stream the data.
I checked the IMU readout, and by reading the yarp port, I noticed that the magnetometer value is constant and equal to zero. 

This behavior is observed in all iCub FT imus. Please notice that iCubGenova09 is currently mounting the FT45 in the feet and the FT58 in the legs.  

This is the `left_leg imu` (FT58).

```
icub@icub-console-xps:~$ yarp read ... /icub/left_leg/imu/measures:o 
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.0.0.150:10004/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /icub/left_leg/imu/measures:o to /tmp/port/1 using tcp
(((-0.125 -0.4375 -0.5625) 1674054506.32432508469)) (((1.27000000000000001776 -0.340000000000000024425 -9.6300000000000007816) 1674054506.32432484627)) (((0.0 0.0 0.0) 1674054506.32432508469)) (((-178.0625 -7.125 100.875) 1674054506.32432508469)) () () () () () ()
```
while this is the `left_foot imu` (FT45) (currently on iCubGenova09 there are two FTs connected to the same ems for the foot)

```
icub@icub-console-xps:~$ yarp read ... /icub/left_foot/imu/measures:o 
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.0.0.150:10004/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /icub/left_foot/imu/measures:o to /tmp/port/1 using tcp
(((-0.125 -0.125 0.0) 1674054603.7856593132) ((-0.25 0.0 -0.1875) 1674054603.78365850449)) (((0.530000000000000026645 -1.1399999999999999023 9.69999999999999928946) 1674054603.7856593132) ((0.770000000000000017764 -1.14999999999999991118 9.64000000000000056843) 1674054603.78365850449)) (((0.0 0.0 0.0) 1674054603.78766036034) ((0.0 0.0 0.0) 1674054603.78365850449)) (((-6.125 -2.875 186.6875) 1674054603.7856593132) ((-6.3125 -4.1875 169.25) 1674054603.7836587429)) () () () () () ()
```

Accordingly to the [`multipleAnalogSensorsSerializations.thrift`](https://github.com/robotology/yarp/blob/ea6ed180ddb102679d8b8492fc40329719d7b8e4/src/devices/multipleAnalogSensorsMsgs/multipleAnalogSensorsSerializations.thrift), the magnetometer is the third element of the yarp bottle.
So for the foot FTs (FT45) we have
```
(((0.0 0.0 0.0) 1674054603.78766036034) ((0.0 0.0 0.0) 1674054603.78365850449))
```
Similarly for the leg (FT58) we have
```
(((0.0 0.0 0.0) 1674054506.32432508469))
```

I also tried to check the magnetometer data coming from the IMU mounted on the iCub head and in this case the value changes as shown in the following plot 

[image]

Here the outcome of the head inertials port
```
icub@icub-console-xps:~$ yarp read ... /icub/head/inertials/measures:o 
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.0.0.150:10004/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /icub/head/inertials/measures:o to /tmp/port/1 using tcp
(((-0.1875 0.1875 0.0) 1674057757.53589510918)) (((0.359999999999999986677 0.800000000000000044409 -9.74000000000000021316) 1674057757.53589510918)) (((3.750000000000000095e-06 3.67499999999999988134e-05 4.89999999999999984179e-05) 1674057757.53589510918)) (((242.375 2.0 -175.3125) 1674057757.53589510918)) () () () () () ()
```

### Additional context

I opened the issue here however I think this is not a problem related to the Hardware of the robot since is involving all the FTs.

@pattacini if you want we can move the discussion in a separate repo. 

cc @traversaro and @G-Cervettini

pattacini, 2023-01-18:
Thanks for reporting @GiulioRomualdi 👍🏻 

>I opened the issue here however I think this is not a problem related to the Hardware of the robot since is involving all the FTs.
>
>@pattacini if you want we can move the discussion in a separate repo.

True that it might be unrelated to the HW but it's still a problem whose progress can be shared with the community herein.

We will plan for it soon.
Stay tuned.

AntonioConsilvio, 2023-01-24:
Hi @GiulioRomualdi, we are organizing to address this issue in early February.

cc @maggia80 @Fabrizio69 @pattacini

pattacini, 2023-01-24:
/remind on February 06

GiulioRomualdi, 2023-01-24:
That's great! Thank you!

pattacini, 2023-02-07:
Just planned for this sprint!

Nicogene, 2023-02-14:
We added https://github.com/icub-tech-iit/ft-sensors-simulink/pull/11 the possibility to read the magnetometer and `IMU_STATUS` data, and although it states that the calibration of the magnetometer is good also with the Simulink model we have only 0 on the magnetometer data

pattacini, 2023-02-14:
Next steps from @Nicogene:

>Since the firmware that handles the imu is shared between the `strain2` and `rfe` (they both mount imu bosch bno055), an interesting test would be to read the rfe magnetometer data and the eul angles.
>
>My idea is that the magnetometer problems lead to jumps on the yaw.
>
>We could try:
>
>- Instead of getting all the data we should try to get only the magnetometer
>- Wild guess: maybe it is a Faraday cage problem induced by the metal of the FT? The test on the rfe should answer to this question. (Tested with no success ❌)
>- Check if there is a problem in the electornics (see [this similar problem](https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BNO055-gyro-0-for-all-values/td-p/8725)) cc @icub-tech-iit/silo-electr

Nicogene, 2023-02-17:
I tried to read the magnetometer data through the old imu we used to mount on icub-head, that has bno055 sensor as well.

[image]

I wrote a very simple executable that instantiates the `multipleanalogsensorsclient` and I got only magnetometer data, and in this case change and it is not 0.

`imuBosch_BNO055` device reads 32 byte starting from the ACC register (0x08)
https://github.com/robotology/yarp/blob/e56e474fcfb574e39945a4c1dbf3256cfad202e6/src/devices/imuBosch_BNO055/imuBosch_BNO055.cpp#L622

The imu is configured in `NDOF`
[image]

In our firmware instead, we are reading 47 bytes, but I don't think this can affect somehow the magnetometer readings.

[image]

And also in the firmware we are setting the `NDOF` mode.

https://github.com/robotology/icub-firmware/blob/e858dc3ab33f6c3f8aeb1f24413f7ef3f50e90d8/emBODY/eBcode/arch-arm/embot/app/embot_app_application_theIMU.cpp#L530

The only difference seems to be the electronics boards.

pattacini, 2023-02-17:
The next step will foresee configuring the FT-IMU in `AMGEQ` mode.

pattacini, 2023-02-23:
Some interesting findings we came up with lately.

From @marcoaccame:

>I have tested the to acquire w/ `Set::AMGEQL` rather than `Set::FULL` and nothing changes.
>
>Moreover, as I spotted that the use of fusion may change the reading of the magnetometer, I did some tests on some boards I have:
>- on strain2: the magnetometer gives zero in both fusion and non fusion mode;
>- on mtb4c: sometimes the magnetometer appears but only if fusion is off.
>- on rfe: magnetometer ok if fusion is on, all zero if fusion is off.
>
>Very strange.

Also, @AntonioConsilvio and @Nicogene reported that, among the three boards, the IMU unit is plugged with a different kit of resistances, which is worth investigating.

Nicogene, 2023-03-13:
We repeated the tests on the 3 boards (`strain2`, `rfe`, `mtb4`) that mount the bno055 imu, and we checked also the sw version of the unit, and it is the same in all the 3 boards, it seems that the magnetometer works only on `rfe`, but it is not clear why.

|   | FUSION  | NO-FUSION   | Bn055 sw version  | 
|---|---|---|---|
| `mtb4`  | ❌  |  ❌  | swRev:785 blVer: 21  |   
|  `rfe`  | ✅   | ✅  | swRev:785 blVer: 21  |  
| `strain2`  | ❌  | ❌   | swRev:785 blVer: 21  |

Fabrizio69, 2023-03-16:
Ciao @Nicogene can you recover code, revision and serial number for the boards under test?

pattacini, 2023-03-16:
>Also, @AntonioConsilvio and @Nicogene reported that, among the three boards, the IMU unit is plugged with a different kit of resistances, which is worth investigating.

Just for completeness, we verified that the differences were relative just to the pull-up stage.
The wiring and the connections are correct in the diagrams. We will be checking if we can spot differences in the boards that are not reported in the schemes.

maggia80, 2023-03-16:
since in the schematic, the only difference between RFE and the other two cards is in the selector pin `BNO055_SELCOM `
Can we check if we are controlling properly this pins setting it to 0? We can check with the oscilloscope to double check. 
[image]

I have also a curiosity, are we using the configuration for using the external oscillator or the internal one? How the [SYS_TRIGGER](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf) is configured?

Nicogene, 2023-03-16:
> I have also a curiosity, are we using the configuration for using the external oscillator or the internal one? How the [SYS_TRIGGER](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf) is configured?

From [this search](https://github.com/search?q=repo%3Arobotology%2Ficub-firmware++eo_sens_bno055_sendWriteCommand%28eo_sens_bno055_register_SYS_TRIGGER%2C+%26data%2C+numOfBytes%29%3B&type=code) it seems that on mtb4 and strain2 is configured the external oscillator, **on rfe it doesn't** (see [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/libs/lowlevel/stm32hal/cubemx/rfe/Src/BNO055.c))
Maybe setting the external clock makes the magnetometer stop working? 🤔

maggia80, 2023-03-16:
I'll change to internal also the RFE to see if this is  the problem

marcoaccame, 2023-03-16:
> > I have also a curiosity, are we using the configuration for using the external oscillator or the internal one? How the [SYS_TRIGGER](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf) is configured?
> 
> 
> 
> From [this search](https://github.com/search?q=repo%3Arobotology%2Ficub-firmware++eo_sens_bno055_sendWriteCommand%28eo_sens_bno055_register_SYS_TRIGGER%2C+%26data%2C+numOfBytes%29%3B&type=code) it seems that on mtb4 and strain2 is configured the external oscillator, **on rfe it doesn't** (see [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/libs/lowlevel/stm32hal/cubemx/rfe/Src/BNO055.c))
> 
> Maybe setting the external clock makes the magnetometer stop working? 🤔 
> 
> 
> 
> cc @marcoaccame 
> 
> 

Hi, this is not the code running on the boards, rather the original HW test project. You have to look at the code inside the embot::hw::bno055 driver.

marcoaccame, 2023-03-16:
> I have also a curiosity, are we using the configuration for using the external oscillator or the internal one? How the [SYS_TRIGGER](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf) is configured?

All boards use the default mode after bootstrap which is internal oscillator.

marcoaccame, 2023-03-16:
> since in the schematic, the only difference between RFE and the other two cards is in the selector pin `BNO055_SELCOM `
> 
> Can we check if we are controlling properly this pins setting it to 0? We can check with the oscilloscope to double check. 

I believe this pin was added to change comm from i2c to usart.

The pin `BNO055_SELCOM` is not driven in the code. The symbol does not even appears in cube mx. 

But the value of PS1, the associated pin on the bno055, must be zero because i2c communicates fine.

marcoaccame, 2023-03-16:
In `embot_hw_bno055.h` add register value `0x3F`

```c++
...

        OPR_MODE        = 0x3D,
        PWR_MODE        = 0x3E,
        SYS_TRIGGER = 0x3F,
        AXIS_MAP_CONFIG = 0x41,
        AXIS_MAP_SIGN   = 0x42
    };

```

In it’s cpp write 0x80 to the above register inside the `init()` function just after the bootstrap of the chip.

````C++
...

s_powerON(s, PORtime);

// enable the external oscillator by writing 0x80 in reg 0x3F
embot::hw::bno055::write(s, embot::hw::bno055::Register::SYS_TRIGGER, 0x80, 5*embot::core::time1millisec);

...

```

That should be all.

pattacini, 2023-03-22:
Hi there,

We have an update on this.

From @marcoaccame:

>I spotted that the mtb4 runs at 16 MHz and the rfe runs at 80 MHz. So I clocked the mtb4 at 64 MHz and bingo!
>
>Just WIP however because at 64 MHz CAN timing are not correct yets.

Stay tuned.

pattacini, 2023-03-23:
@marcoaccame has churned out a patch in:
- https://github.com/robotology/icub-firmware/pull/359

We need to thoroughly test it and be careful while merging as there are other pending PR's in the queue.

pattacini, 2023-03-29:
@marcoaccame drafted a PR:

- https://github.com/robotology/icub-firmware/pull/361

Nicogene, 2023-03-30:
https://github.com/robotology/icub-firmware/pull/361 fixed indeed the magnetometer, see the plots above

Thanks heap @marcoaccame!

## Magnetometer
### `strain2`
[image]

### `mtb4`

[image]

cc @GiulioRomualdi

GiulioRomualdi, 2023-03-30:
That's great! Thank you a lot for the effort!

marcoaccame, 2023-03-30:
OK, thanks @Nicogene. I have just merged the PR for the sources:
- https://github.com/robotology/icub-firmware/pull/361

As I understand you will take care of the PR for the binaries which will soon follow. 

In the meantime, I attach in here the new binary v2.1.0 for the `strain2` board: [strain2-v2.1.0.zip](https://github.com/robotology/icub-tech-support/files/11110540/strain2-v2.1.0.zip)

Nicogene, 2023-03-31:
The fix has been delivered also in terms of binaries:
- https://github.com/robotology/icub-firmware-build/pull/77

I think we can close the issue 🚀 

cc @GiulioRomualdi

closed 2023-03-31 by Nicogene