robotology/mesh-support #1486: iCubGenova11 - One of the neck belts seems damaged

xEnVrE, 2023-01-19:
### Device name 🤖

iCubGenova11

### Request/Failure description

We noticed that one of the neck belts seems damaged. See the following picture.

[image]

We were wondering whether it is the case to substitute it.

### Detailed context

We noticed that one of the neck belts seems damaged. See the following picture.

[image]

We were wondering whether it is the case to substitute it.

AntonioConsilvio, 2023-01-19:
Hi @xEnVrE, me and @GiacomoVassallo13 replaced the damaged neck belt with a new one. We tried it and it seems work without any problem. 

Should the new belt cause problems, you can reopen this issue.

cc @Fabrizio69

closed 2023-01-19 by AntonioConsilvio

xEnVrE, 2023-01-19:
Thanks @AntonioConsilvio @GiacomoVassallo13