robotology/mesh-support #1488: iCubGenova11 - The robot oscillates abruptly every so often

xEnVrE, 2023-01-20:
### Device name 🤖

iCubGenova11

### Request/Failure description

This issue is to report that sometimes, but rarely enough to get a video of the phenomenon, the robot has strange oscillations spikes that last 1/2 seconds. When they happen I see the arms oscillating on the horizontal plane hence I am assuming that the torso yaw is oscillating.

The oscillations is not negligible and I think we should for sure understand why it happens and avoid it.

I am attaching the log file in the hope that there are some warnings or errors that could explain the fact.

Also @Nicogene noticed it while using the robot.

### Detailed context

This issue is to report that sometimes, but rarely enough to get a video of the phenomenon, the robot has strange oscillations spikes that last 1/2 seconds. When they happen I see the arms oscillating on the horizontal plane hence I am assuming that the torso yaw is oscillating.

The oscillations is not negligible and I think we should for sure understand why it happens and avoid it.

I am attaching the log file in the hope that there are some warnings or errors that could explain the fact.

### Additional context

[log_icub-head_yarprobotinterface_1436.txt](https://github.com/robotology/icub-tech-support/files/10467307/log_icub-head_yarprobotinterface_1436.txt)

Nicogene, 2023-01-24:
Today I worked on `iCubGenova11` and the oscillation didn't happen this time.

But I got a lot of these errors
```
from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 815s 330m 861u: (code 0x0200000d, par16 0x0003 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```
It is suspicious because it is the right arm that oscillates

AntonioConsilvio, 2023-01-24:
Hi @xEnVrE I have encountered this problem, I will do some tests to understand the causes of this strange behavior.

cc @Fabrizio69

xEnVrE, 2023-01-24:
As discussed f2f with @AntonioConsilvio, we would like to postpone such tests after 30/01/2023 due to a demo with this robot.

Thanks

xEnVrE, 2023-01-27:
We finally got a video of the thing (the relevant part is repeated several times):

https://user-images.githubusercontent.com/6014499/215065758-b0f065c5-29ae-468f-9a1c-5befdcd09a42.mp4

It seems to be the `r_shoulder_yaw` then.

cc @AntonioConsilvio

maggia80, 2023-01-27:
yes, it's the `r_shoulder_yaw`. Two options: the AEA is not well connected (we should plot the encoder readings), the phases of the motor are not well connected.

xEnVrE, 2023-02-01:
Hi @AntonioConsilvio,

Do you think is it possible to investigate this next week? Thanks

AntonioConsilvio, 2023-02-02:
Hi @xEnVrE, this is possible, I think on Monday or Tuesday.

cc @sgiraz

AntonioConsilvio, 2023-02-06:
Hi @xEnVrE this problem should be solved. I found a cable on the Yaw's AEA sensor to be loose. I have tried this joint several times and it seems to work without any problems.

If the problem recurs, you can reopen this issue

@sgiraz

closed 2023-02-06 by AntonioConsilvio

xEnVrE, 2023-02-06:
Many thanks @AntonioConsilvio