robotology/mesh-support #1489: Strain 2 not detected on EB6 (left leg) at startup

gabrielenava, 2023-01-20:
### Device name 🤖

iCubGenova04

### Request/Failure description

Strain 2 on the left leg is not detected at startup, and the robot does not start

### Detailed context

We removed the covers of the iRonCub lower body to mount the protective pants from ITAI. The robot was working before the operation, and it was not starting anymore after. I think that the FT cable got disconnected from the crimp during the "pants wearing" operation.

### Additional context

cc @vpunithreddy @HosameldinMohamed 

### How does it affect you?

we are planning an experiment next week

HosameldinMohamed, 2023-01-20:
@AntonioConsilvio came and fixed the broken cable, now the robot is luanched successfully 🎉 

The broken wire is the black one from the cable `6N2` connecting the board `EB6` with the strain2 of left leg.

@AntonioConsilvio said he'll pass again to make the wires of the cable even.

AntonioConsilvio, 2023-01-24:
Hi @gabrielenava @HosameldinMohamed, this morning I changed the molex of the 6N2 cable. 

I am waiting for feedback on the operation of the robot.

cc @Fabrizio69

closed 2023-01-24 by AntonioConsilvio