robotology/mesh-support #1490: Leg Cabling: Replacing the cable 14

jhielson, 2023-01-26:
### Device name 🤖

iCub[Edinburgh][Edinburgh Centre for Robotics], S/N[034]

### Request/Failure description

Hi everyone,

I need help replacing a cable in the iCub robot. The crimp part of the cable 14 snapped today during an experiment. 

### Detailed context

Following your documentation, there is a white cover that needs to be removed to access the cable. In my robot, I can see the white cover but the free space between the legs (Photo 26: https://icub-tech-iit.github.io/documentation/icub_tendons/waist_legs/#31-hip-flexion-cable-14 ) does not exist. There are a lot of cables and a metallic part with the tag OB4 in this area. Can you please give me some guidance on how to remove those parts so I can replace the cable? I could not find the instruction on the iCub documentation.

### Additional context

[image]

AntonioConsilvio, 2023-01-30:
Hi @jhielson, the picture you can see in the guide is without electronics boards. Now, to gain enough space to be able to replace the steel cable, you can try removing the waist cover (green):

[image]

Then you need to remove the metallic part by removing the 4 screws (orange):

[image]

Two boards are attached to the metallic part, so when you remove it you should have a view similar to this:

[image]

However, you need to be very, very careful because in these photos taken by CAD, there are no electrical cables. So when you remove the metallic part, it should remain hanging because it should be held by the electrical cables.

But the electrical cables are very, very delicate, so it's possible that during this procedure one or more of the cables could come loose, and so you have to be very careful.

I'm not entirely sure that you can gain enough space to replace the cable while the metal part is hanging, but you can try to perform these steps and see.

If you cannot get enough space, you need to unplug the electric cables and remove the metal part. But before you do that, I recommend that you write to us because we may provide you with some useful information.

If you have any questions, write to us.

cc @Fabrizio69

jhielson, 2023-02-01:
Hi @AntonioConsilvio,

Thank you. The problem is on the cable 14 that is located in the left leg of the iCub. After removing the robot's back protection and all the four screws, I could see the two tiny ones holding the white cover. I have been trying to remove them but it is really hard to reach them.

jhielson, 2023-02-01:
Hi @AntonioConsilvio,

I just managed to remove the cover and completely remove the old cable but I am not sure how to put the new one. Do you have any tips for that part.

jhielson, 2023-02-02:
Hi, It was a bit challenging but I managed to replace the cable. The leg is working fine now. Thank you very much!

There is another thing happening to the robot that I would like to discuss in this repo but I will open a new issue for that.

AntonioConsilvio, 2023-02-02:
Hi @jhielson good job because cable replacement is not easy.

Now, I have a few suggestions for you.

- When you start the robot the torso joints automatically go to idle, is that right? If so, you can check the terminal or logger to see if `yarprobotinterface` sends any messages. 
- Then, when these joints are in idle, you can try moving them manually and see if the encoder value changes and by how much. 
- After these steps, if everything looks fine, you can press on `yarpmotorgui` the "run" button and the torso should work. (Keep the red button close when performing this action).
- If the torso starts without any problems, try moving it from `yarpmotorgui` and check how it works.
- If everything seems fine, try restarting the robot and see if the problem persists.

Then, send us feedback on all these steps. 

If you have any questions, write to us.

cc @sgiraz

jhielson, 2023-02-02:
Hi @AntonioConsilvio,

Thank you for your support. I just checked the encoders here and they seem to be updating the values fine every time I move the three joints manually. Below, I filtered the error messages when I executed ```yarprobotinterface``` and it seems the torso "thinks" the fault button is pressed, which is not. Do you know how to fix that?

[image]

AntonioConsilvio, 2023-02-02:
@jhielson I may have figured out the problem, so I am sending you a pdf file.
[iCub2.5-E3.3.2_5672_Logic.pdf](https://github.com/robotology/icub-tech-support/files/10569341/iCub2.5-E3.3.2_5672_Logic.pdf)

This is the wiring diagram of the robot. 
So if you go to page 9 (which should be about waist, you can read it on the page in the lower right corner), and if with ctrl+f you search for "fault," on this page you should find about 6 white wires connecting the fault when you press the red button. 

Probably one of these cables has been detached when replacing the steel cable, and the board receiving this cable "thinks" that the red button was pressed, even though it was not.

So one (or more) of these cables has come loose and you need to find it in the waist and repair it.
Maybe one thing can help you: the board that "feels" the disconnected cable should have a red LED (I'm not sure, but it should).

Another piece of information that you might find useful: the "2FOC" boards you can read about in the pdf are the ones with the big black heatsink.
While the EMS does not have it.

If you have any questions, write to us

cc @sgiraz

jhielson, 2023-02-02:
Hi,

It looks like the cables are not loose. I checked from both sides. Is there another way to test them?

[image]
 
[image]

AntonioConsilvio, 2023-02-06:
Hi @jhielson, in the picture you sent me you are checking the wrong board. The white cable you need to check is connected to the boards called 2FOC, and they are like this one:

[image]

With a black heatsink.

For example, in your photo you can see the white cable in question:

[image]

The white cable visible in the photo needs to be checked at all its branches going to the waist (and which should be connected to the 2FOCs) and check that it has not been cut by a mechanical part or that it is disconnected.

If you have any question, write to us.

cc @sgiraz

AntonioConsilvio, 2023-02-06:
@jhielson In the picture below you can see an example of how the fault cable is connected to a 2FOC board:

[image]

This is not the 2FOC in the robot's waist, but it is just to show that the fault cable is not in the same sheath with the other cables, but should come from another sheath and connect to the motor connector.

It can also be seen in the PDF I sent you. I take only one connector as an example, but it is the same for the other 3 connectors of the 3 torso motors:
 

[image]

cc @sgiraz

jhielson, 2023-02-06:
Thanks for the update. I noticed that I was looking to the wrong board last Friday. Unfortunately, the correct board is attached to the metallic part. I am gonna try to separate those parts. Otherwise, I can not see the connection of one of the white cables. The other two seem to be fine. I used a multimeter to test the continuity of the wires, and they seemed to be fine.

jhielson, 2023-02-06:
Sorry, I was using my phone to leave the last comment,  and I closed the issue by mistake.

jhielson, 2023-02-06:
Hi,

It seems the board is also glued to the metallic part. I removed the four rubber screws but it did not work. The visibility is quite bad. 

[image]

jhielson, 2023-02-08:
Hi, 

I believe the issue might be in the 0B3/2FOC board, more specifically at the the white cable that is connected to the component P6_3_FAULT/HALL 0.  As you can see below, it is not easy to reach that part. I wonder if there is an easy way of removing this board from the metallic part.

[image]

[image]

jhielson, 2023-02-09:
Thanks @AntonioConsilvio, I have already removed those 4 screws but the black heatsink is glued to the metallic part. I am trying to remove but it is not working.

[image]
[image]

AntonioConsilvio, 2023-02-13:
Hi @jhielson, there is this a conductive sheet that makes adhesive effect between the board and a metallic part.

But you should be able to disconnect the connector without unplugging the 2FOC board by following this video:

https://user-images.githubusercontent.com/114915464/218449706-8ee58d8e-61bb-4830-b59e-fa002ed5ab48.mp4

This is just a simulation, I know you have a lot of extra cables and other things that can make this procedure more difficult, but with a little effort you should be able to disconnect the connector with a flat-blade screwdriver (in the video I only pry from the right side, but you can help by prying from the left and right sides).

So, to reconnect the connector, you can do this:

https://user-images.githubusercontent.com/114915464/218450819-c7157fa2-6347-4b23-9b4c-9f9b44656f6f.mp4

With tweezers you can pick up the connector from the base and connect it to the board.

These steps are not easy but should be possible. 

If you have any questions, write to us.

cc @sgiraz

jhielson, 2023-02-13:
Hi @AntonioConsilvio,

Thank you for your suggestion. I managed to remove the connectors and test them. They all seem to be working fine. I tested the wire continuity between each white cables (Photo 01-03) and the one for the right leg (Photo 04). The multimeter beeped for every single one. 

Photo 01: Torso
[image]

Photo 02: Torso
[image]

Photo 03: Torso
[image]

Photo 04: Right Leg
[image]

AntonioConsilvio, 2023-02-20:
Hi @jhielson, we consulted and came to the conclusion that the problem should reside in those cables there. So if you are sure that with the multimeter seems everything fine, you can try to test the crimp of the cable with a tweezers, as in this video:

https://user-images.githubusercontent.com/114915464/220156198-f5cda52b-bf6d-4e8a-826e-5e10f4ec9893.mp4

You can try pulling the cable with tweezers, without tearing it, just to see if the crimp still holds.

If, after this test, the crimps look fine, you can try connecting the motors again and retry the `yarpmotorinterface` to see if the problem persists. 

If the problem persists, we can schedule a videocall to provide you with better support.

cc @sgiraz

jhielson, 2023-02-21:
Hi @AntonioConsilvio,

I pulled a litle bit the white cable of each connector and they seem to be fine. I also made a video testing the wire connectivity of those cables. My hands are not too stable but I think you can see the process. 

https://user-images.githubusercontent.com/2223912/220374223-2c984a26-76b4-42c0-aad4-6e9a04af9e31.mp4

https://user-images.githubusercontent.com/2223912/220374241-a57a3a18-9a7b-45d7-a2a6-45fc5687db60.mp4

https://user-images.githubusercontent.com/2223912/220374261-3d3f5561-c887-428e-a3dc-82342dd7227b.mp4

I copied here some error messages  that I got during the execution of ```yarprobotinterface```:

```

587	1,065038	ERROR	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 401s 270m 97u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
588	1,065084	ERROR	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 401s 270m 185u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
589	1,065100	ERROR	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 401s 270m 283u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
615	1,106314	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 401s 115m 730u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
619	1,115688	ERROR	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 401s 319m 140u: (code 0x02000008, par16 0x0002 par64 0x0000000000000002) -&gt; MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
652	1,328448	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 401s 328m 902u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
653	1,328558	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 401s 328m 999u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
654	1,328595	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 401s 329m 106u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
655	1,328624	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 401s 329m 197u: (code 0x02000000, par16 0x0003 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
690	1,723672	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 401s 736m 231u: (code 0x02000010, par16 0x0000 par64 0x0000000000000000) -&gt; MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
691	1,723781	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 401s 736m 319u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
692	1,723817	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 401s 736m 417u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
709	1,733947	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 401s 739m 234u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -&gt; MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
710	1,733980	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 401s 739m 353u: (code 0x02000010, par16 0x0002 par64 0x0000000000000000) -&gt; MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
711	1,734009	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 401s 739m 479u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -&gt; MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
724	1,804956	ERROR	 from BOARD 10.0.1.23 (face-eb23-j2_5), src LOCAL, adr 0, time 401s 816m 970u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
725	1,805057	ERROR	 from BOARD 10.0.1.23 (face-eb23-j2_5), src LOCAL, adr 0, time 401s 817m 69u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
726	1,805094	ERROR	 from BOARD 10.0.1.23 (face-eb23-j2_5), src LOCAL, adr 0, time 401s 817m 172u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
727	1,805123	ERROR	 from BOARD 10.0.1.23 (face-eb23-j2_5), src LOCAL, adr 0, time 401s 817m 269u: (code 0x02000000, par16 0x0003 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
728	1,926731	ERROR	 from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 401s 941m 745u: (code 0x02000010, par16 0x0000 par64 0x0000000000000000) -&gt; MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
729	1,926833	ERROR	 from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 401s 941m 898u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -&gt; MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
[... cut ...]
1160	2,838968	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 402s 854m 227u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
1161	2,848624	ERROR	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 402s 854m 328u: (code 0x02000000, par16 0x0003 par64 0x0000000000000000) -&gt; MC: exernal fault button pressed. + .
1162	2,848691	ERROR	 from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src CAN1, adr 0, time 402s 853m 108u: (code 0x00000019, par16 0x2d01 par64 0x0000000000000000) -&gt; SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .

```

sgiraz, 2023-02-21:
Hi @jhielson,

Looking at your log I see that:

| board       | hard limit reached | external fault button pressed | AEA encoder invalid data | could not tx frames on CAN bus |
| :---------: | :-------------------: | :------------------------------: | :---------------------------: | :----------------------------------: |
| 10.0.1.5    |                               |                      X                       |                X                         |                                                     |
| 10.0.1.21  |                               |                      X                       |                                           |                                                     |
| 10.0.1.22  |                               |                      X                       |                                           |                                                     |
| 10.0.1.23  |                               |                      X                       |                                           |                                                     |
| 10.0.1.25  |              X               |                      X                       |                                           |                                                     |
| 10.0.1.26  |              X               |                                               |                                           |                                                     |
| 10.0.1.27  |                               |                                               |                                           |                           X                        |
| 10.0.1.28  |              X               |                                               |                                           |                                                     |

All the parts involved in the errors reported have been highlighted in the following schematic:

Where:
- green (🟢) --> AEA encoder invalid data
- red (🔴)     --> external fault button pressed
- purple (🟣) --> hard limit reached
- blue(🔵)    --> could not tx frames on CAN bus

[image]

These errors indicate that there are many problems in addition to the external fault, so we would appreciate if you could send us the full log of the `yarprobotinterface` after running the [whole robot][1] from a fresh start.

**Notes:**
- In the meanwhile, we gonna inspect for further problems by looking at the data provided by you

cc @AntonioConsilvio 

[1]: https://github.com/robotology/robots-configuration/blob/master/iCubEdinburgh01/icub_all.xml

jhielson, 2023-02-21:
Hi,

You can find below a full log file after running yarprobotinterface.

[yarprunlog_21_02_2023_16_37_29.log](https://github.com/robotology/icub-tech-support/files/10795673/yarprunlog_21_02_2023_16_37_29.log)

AntonioConsilvio, 2023-02-22:
Hi @jhielson, you should use the `FirmwareUpdater` command on the iCub head (with -X because it is a graphical application), then select the first option and click Discover. Here you will be able to see all the electronic boards in your robot, so send us a screenshot of what you see.

cc @sgiraz

jhielson, 2023-02-22:
Hi @AntonioConsilvio ,

It seems I can not initiate the drivers via FirmwareUpdater. But the robot was on and connected.

[image]

[image]

jhielson, 2023-03-07:
Hi,

I just gotta back from sick leave. I am not sure if it is a good idea to open the other parts of the robot to check those connectors. They are located in places that are difficult to reach and the cables are quite delicate. But I can do it if those connectors might represent the real issue.

sgiraz, 2023-03-09:
Hi @jhielson,

In our internal comparison analysis, based on your logs, we suspect that there may be many different kinds of issues that need to be resolved. We would like to ask you to send us the robot so that we can fix all of the problems and make it work again.

If you agree, here is **the address where to send the parcel** for the maintenance activity:

IIT - Istituto Italiano di Tecnologia
Via Morego 30
I-16163 Genoa, ITALY
Att.n: Fabrizio Larosa (iCub Tech, CIRS)
Phone: +39 010 28 98 223

**⚠️ IMPORTANT**: please safely pack the goods using the maximum care to absorb shocks. If possible, indicate "fragile" on the parcel. As soon as you have a tracking number, pls let us know so we may follow the shipping.

@andreaderito, please add further relevant info if needed.

cc @Fabrizio69 @maggia80 @AntonioConsilvio

jhielson, 2023-03-09:
Hi,

I appreciate all the support given to me to solve the issue. I do not think the project that I have been working on has a budget to send the robot for repairing. I need to talk to my supervisors. But I was wondering, except by the three joints at the torso, the robot seems to be working fine (video below). There are some minor issues that we do not mind right now, like the tactile sensors failing in the right arm. I still believe the fault might be in those boards that we have discussed since they control the three joints.  

https://user-images.githubusercontent.com/2223912/224004496-2c136ccf-1e2a-4f4d-a606-6c7214e98e72.mp4

sgiraz, 2023-03-22:
Hi @jhielson, 

We may schedule a video call for the next week and try to solve the problems together.
Stay tuned I'll give you some possible dates soon.

Anyway, if we'll not able to solve the issues in this way, we would encourage you to send the robot for better checkup and maintenance.

cc @AntonioConsilvio @Fabrizio69

sgiraz, 2023-03-28:
Hi @jhielson, 

If you agree, we can schedule the video call to support this activity on Thursday, March 30 at 10:00 (GMT+2).
Next Monday, April 3 at 10:00 (GMT+2) is another possibility.
Please let us know what date works best for you.

cc @AntonioConsilvio @Fabrizio69

jhielson, 2023-03-28:
Hi @sgiraz,

I will be at the University of Manchester on Thursday. But I can do it on Monday.  Thank you!!

sgiraz, 2023-03-31:
Hi @jhielson,

## What you should have before the meeting is: 
- Digital Multimeter
- Robot Laptop
- Screwdrivers (different sizes and kinds)
- Solder[^1]
- Tweezers
- Crimping tool
- Crimp female AWG 26-28
- Crimp female AWG 28-32
- Scissor / wire stripper
 
**Notes**: With high probability, we need to see the physical robot. For this purpose, we ask you to use an additional external camera connected to your laptop (or you may a mobile phone camera and switch the video call on it) 

[^1]: optional items

jhielson, 2023-04-03:
Hi,

Good morning. Which link should I use for the meeting?

sgiraz, 2023-04-03:
> Hi,
> 
> Good morning. Which link should I use for the meeting?

Hi @jhielson, I sent you the invitation link to your personal mail.

jhielson, 2023-04-03:
Hi,

I haven't received the email yet. Can you send it to jm2147@hw.ac.uk?

sgiraz, 2023-04-03:
Hi @jhielson,

We have news for you! 📢

---

If you are on time, please do not perform the 2 action points 🎯 suggested during the last call, because they may not be necessary!

---

On the `2FOC` board there is a 100 Ω resistor close to the 3rd pin of the connector that includes the fault wire (⚪). 
Since we saw that the continuity tests on the fault white wires  (⚪) seem fine, on both `2FOC` boards, it is reasonable to suspect that this resistor is broken.

The images below show you the steps to check for this issue:

1. Remove the `2FOC` boards (labeled with `0B3` and `0B4` ) from the robot, and put them on your desk. 💡To remove the glued part you may try to prehead the metallic plate (⚠️ be careful to not overheat the board) and leverage it with the help of a screwdriver.

2. Remove the black heatsink from the `2FOC` boards
[image]

3. Pay attention to these resistors.
[image]

4. Test the value of the resistors using the multimeter. **They must be ~100 Ω each, otherwise, they have to be replaced! 🚨**
[image]

let us know if you were able to do this test and the result you got in step `4`.

cc @Fabrizio69 @AntonioConsilvio

jhielson, 2023-04-04:
Hi,

The ones from the back side of board are working. Can we use them instead?

[image]

sgiraz, 2023-04-05:
Hi @jhielson, 

Unfortunately, as you can see from [the electrical scheme][1], the only connectors that manage the fault signal are `P6` and `P1`. Hence, the connectors are now swappable.
However, to move on you can do the following things in order:
1. remove the faulted resistors and close the circuit soldering a drop of tin
2. now test if the torso works correctly
3. ✅ if the torso now works you can **choose one of the following** action points 🎯:
  - continue to use the robot leaving the drop of tin soldered at point 1, at your own risk ⚠️
  - remove the drop of tin placed at point 1 and solder three brand-new (100 Ω) resistors in place of the previously faulted resistors
  - remove the drop of tin placed at point 1, remove three (100 Ω) resistors from `P2` connector of `0B4` board, and solder them in place of the previously faulted resistors. This option is possible because the `P2` connector of `0B4` actually should be used at all.
4. ❌ If the torso still not working, the microcontroller of the `2FOC` boards may be damaged seriously.  In this case, you can return them to us and we can ship you 2 brand-new boards.

**Notes:**
These are the three resistors you can bring from `P2` connector as described in one of the possible action points of point 3:
[image]

cc @AntonioConsilvio 

 
[1]: https://github.com/icub-tech-iit/electronics-boards-public/blob/master/motor-control/2foc/pads/SE_2FOC_08.pdf

jhielson, 2023-04-06:
Thanks Simone. 

Next week, I will follow the steps above and keep you updated. Meanwhile, could you send me the quote for new boards? I am going to talk to my supervisors so we can arrange that.

Best regards,
Jhielson M. Pimentel

sgiraz, 2023-04-07:
Hi @jhielson.

The price for a single `2FOC` board is indicatively 300 euros.

cc @andreaderito @maggia80

jhielson, 2023-04-27:
Hi @sgiraz, 

I would like to give you an update. We have sent both boards to our partner here in Edinburgh to replace the resistors. It might take a few days though. I will keep you all updated. Thanks.

jhielson, 2023-10-19:
Hi everyone, 

Today, my colleague and I finally managed to put the recently fixed boards back (0B3 and 0B4) to the robot. Unfortunately, the 3 motors from the robot's waist is not responding yet. We got the log file as you can find bellow. Please, let me know if you have any suggestions. 

Error messages:
[image]

Log file below:
[log_icub-head_yarprobotinterface_1461.txt](https://github.com/robotology/icub-tech-support/files/13044865/log_icub-head_yarprobotinterface_1461.txt)

sgiraz, 2023-10-31:
Hi @jhielson,

From the log posted above I see that the first (periodic) errors are the following:

```shell
 
581 1,165692 <ERROR>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 833s 437m 167u: (code 0x02000003, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC hall sensors fault. Invalid sequence in motor Hall effect sensors, please check motor hall cable connections. par16 = ID of joint. + .
 
582 1,165727 <ERROR>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 833s 438m 155u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
```

Are you sure that the phases of the motors are connected correctly to the one re-mounted 2FOC board?
See the picture below for a comparison:

[image]

jhielson, 2023-11-01:
Thank you for your swift response. Initially, we reviewed the schematic and made adjustments to the cable connections, as outlined below (using the sequence white, red, and black instead of black, red, and white). But, we have also tested the setup according to your recommended configuration. Unfortunately, it did not yield the desired results. 

The hall cables seems to be well connected.

[image]

[image]

[image]

sgiraz, 2023-11-10:
Hi @jhielson,

I may generated a bit of confusion about the wiring of the phases (with my screenshot).
Could you check how the phases of other 2FOC boards mounted on the robot are connected, such as those on the legs?
ps. I need to know the color sequence of the phase's wires or just provide some pictures.

If the connections are all equal, another possibility may be that the 2FOC boards repaired are still damaged. In this case, we may provide you with 2 brand-new 2FOC boards.

jhielson, 2023-11-10:
Hi @sgiraz,

No problem. I can provide you more pictures of both boards, but can you point me in the image below which of those wires are the phase ones? 

[image]

I was wondering if it would be a good idea to swap both boards from the robot's right leg with the ones on the robot's back. That could help us understand if the problem persists in the boards or not. What do you think?

[image]
[image]
[image]

jhielson, 2023-11-17:
Hi @sgiraz,

Good news! Following the exchange of boards 3B9 and 0B3, the three torso joints regained functionality, confirming that the issue is with the 0B3 board. You mentioned the potential for providing one of the boards. Could you please provide guidance on the next steps or how to proceed in addressing this matter? Thank you.

sgiraz, 2023-11-20:
> Good news! Following the exchange of boards 3B9 and 0B3, the three torso joints regained functionality, confirming that the issue is with the 0B3 board
Glad to hear that! 🚀

> Could you please provide guidance on the next steps or how to proceed in addressing this matter? Thank you.

Sure, you can contact @andreaderito (andrea.derito@iit.it) in order to proceed with the purchase process.

andreaderito, 2023-11-20:
Dear @jhielson please send me an e-mail (the offering process will be run via e-mail) kindly indicating:
1) what you wish us to offer in the quotation, and related quantity per item;
2) to whom I should address the quotation (title, name &surname, email address)
3) Name of your institution and address
so we may quickly prepare the quotation for you.
Looking forward to your kind reply - thank you in advance
cc @sgiraz

jhielson, 2023-11-20:
Thank you! We will contact you soon.

jhielson, 2023-11-28:
Hello @sgiraz, I hope this message finds you well. We reached out via email to [andrea.derito@iit.it] last week, but unfortunately, we haven't received a response. Could you please verify if the email address is correct on your end? Your assistance would be greatly appreciated. Thank you!

andreaderito, 2023-11-28:
Hello @jhielson yes, I confirm reception, processing the request asap - thx 4 yr kind waiting and best
cc @sgiraz

andreaderito, 2024-02-02:
Hi @jhielson according to the express courier's tracking website, you should have received the ordered goods on 31.01.2024, may you pls be so kind as to confirm to us reception, at your earliest convenience?
Thx and best, Andrea
cc @sgiraz

jhielson, 2024-02-02:
Hello everyone, we received both boards this week. Today, I replaced the faulty one with one of the new boards, but it didn't work as expected. Normally, the lights on the boards blink green only a few seconds after turning the motors on. However, the new board keeps flashing red and green. I tested both new boards, and they exhibit the same behavior. Can you help me with this issue?

andreaderito, 2024-02-02:
Thank you @jhielson for confirming reception of the 2 boards.
Ok, for the problem you mention, let me cc @sgiraz here to follow-up.
cc @AntonioConsilvio @Fabrizio69 @maggia80

jhielson, 2024-02-07:
Thank you for your quick response. I'll be testing those instructions soon. However, yesterday, one cable became loose. Could you please provide the specifications for the connector used in that cable? Mayve I can make one here.

[image]

That's the 5N1 cable.

AntonioAzocar, 2024-02-08:
Hi @jhielson ! 

The crimps to use can be found:

-  at `Farnell` 'site  at the [link](https://uk.farnell.com/molex/50058-8000/contact-socket-28awg-crimp/dp/1756744?ost=1756744) -  Supplier code: **1756744**

-  or at `RS COMPONENTS S.P.A.` 'site at the [link](https://uk.rs-online.com/web/p/crimp-contacts/6706414?searchId=140d3fdb-9522-40a3-a4ab-0ac332036f50&gb=s)  - Supplier code: **670-6414**

As for the connector to use, you can find it:

- at `Farnell` 'site  at the [link](https://uk.farnell.com/molex/51021-0600/connector-housing-rcpt-6pos/dp/1012261?ost=1012261) - Supplier code: **1012261**
- at `RS COMPONENTS S.P.A.` 'site at the [link](https://uk.rs-online.com/web/p/wire-housings-plugs/2799178?searchId=fe7fe85b-7046-41c7-9026-67b0a562f059&gb=s) - Supplier code: **279-9178**
- or at `MOUSER ELECTRONICS`  at the [link](https://eu.mouser.com/ProductDetail/Molex/51021-0600?qs=ZwgtpdmWYYSz82DH2HMXiQ%3D%3D) - Supplier code: **538-51021-0600**

Also, here you can see the schematic of the `5N1` cable

[image]

 cc @sgiraz

jhielson, 2024-05-02:
Hi everyone, sorry for taking too long to reply. I could only work on those cables this week. I fixed them and, now, I will setup the new board. I will keep you updated on that. Thank you.

jhielson, 2024-05-02:
Hello everyone,

A few weeks ago, I transferred the faulty board from the robot's torso to its right leg. Consequently, the upper body regained functionality, but the right leg ceased to function. 

Today, I replaced the faulty board in the right leg of the robot with a new one. Then, I ran the following commands:

```
$ ssh -X icub@10.0.0.2
$ FirmwareUpdater
```

When I push the discover button, I received an error message as shown in the image below.

[image]
[image]
[image]

AntonioConsilvio, 2024-05-06:
Hi @jhielson! I suggest following the procedure described in this [comment](https://github.com/robotology/icub-tech-support/issues/1490#issuecomment-1926524668) to check and if necessary perform a board update.

So, specifically, in the `FirmwareUpdater` window click the `ETH<eth1>` option, and not `CFW2_CAN<0>`!

jhielson, 2024-05-06:
Hi @AntonioConsilvio, 

Thank you for your response. I'll be out of the office tomorrow, but I'll return on Wednesday. I'll explore the ETH option as soon as I'm back and keep you informed.

jhielson, 2024-05-08:
Hi @AntonioConsilvio ,

Thank you. You can find below the screenshot of the ```FirmwareUpdater``` window. Since I have installed the fault board to the right leg of the robot, I discovered the 10.0.0.8 ems4 board. Different from the left leg (10.0.0.6), it only shows 2 FOC boards. Should I double check the installation and try it again?

[image]

[12 more replies]

closed 2025-05-04 by github-actions[bot]