robotology/mesh-support #1491: Right arm shoulder tendons broke

gabrielenava, 2023-01-27:
### Device name 🤖

iCubGenova04

### Request/Failure description

several right shoulder tendons got broken during an iRonCub experiment

### Detailed context

after switching off the controller, the arm started to oscillate and this lead to the tendons to broke before we were able to fault the robot

DanielePucci, 2023-01-28:
@Fabrizio69 @maggia80 unfortunately, on Monday late afternoon and Wednesday we have two important demos https://github.com/ami-iit/lab-events-demos/issues/236 and https://github.com/ami-iit/lab-events-demos/issues/237 

Let us know you can help with fixing the robot, thanks

AntonioConsilvio, 2023-01-30:
Hi @DanielePucci, we will fix this issue this morning.

cc @Fabrizio69 @gabrielenava @maggia80

AntonioConsilvio, 2023-01-30:
Hi @gabrielenava, @GiacomoVassallo13 and I replaced damaged cables (6, 7 and 8) with three new ones.

I await feedback on the operation of the robot.

cc @Fabrizio69

DanielePucci, 2023-01-30:
Thank you all guys, it was important to have the robot fully working today

gabrielenava, 2023-02-01:
>I await feedback on the operation of the robot.

it is working fine now, thanks @AntonioConsilvio!

closed 2023-02-01 by AntonioConsilvio