robotology/mesh-support #1492: iCubGenova11 - Problem with r_pinky.

fedeceola, 2023-01-30:
### Device name 🤖

iCubGenova11

### Request/Failure description

When closing the `r_pinky` joint the ring and the little don't close synchronously differently from the left hand.

cc @xEnVrE 

### Detailed context

-

AntonioConsilvio, 2023-01-30:
Hi @fedeceola @xEnVrE, the problem was detected in this pulley on the ring of the left hand:
[image]

The cable does not run around the pulley as it should. 

We have arranged with @Nicogene (who has to use the robot tomorrow), and @GiacomoVassallo13 and I will do the intervention tomorrow morning (around 9:30 am).

AntonioConsilvio, 2023-02-01:
Hi @fedeceola @xEnVrE, @GiacomoVassallo13 fixed this problem. Now the cable go around the pulley without problems.

[image]

The problem regarding synchronized closing however persists, but in reverse. Whereas before the ring finger closed earlier than the little finger, now the little finger closes slightly earlier than the ring finger. 
To fix this problem, it is necessary to rewire the ring finger and little finger, however since the wiring process is quite sloppy, it is not guaranteed to fix the problem. It may fix the problem, or it may not. 

So, counting that rewiring requires opening the hand and forearm, do you want to try it anyway or not?

cc @sgiraz

xEnVrE, 2023-02-01:
Thanks for the fix @AntonioConsilvio @GiacomoVassallo13 

Regarding the more invasive maintenance, we will discuss tomorrow and let you know!

AntonioConsilvio, 2023-02-06:
Hi @xEnVrE @fedeceola, maintenance is not needed for now, so I am closing this issue.
However, if you find this maintenance necessary in the coming days, you can reopen this issue or open a new one.

cc @sgiraz

closed 2023-02-06 by AntonioConsilvio