robotology/mesh-support #1493: Right thumb Tactile sensor not responding

Ryo-Naohara, 2023-01-30:
### Device name 🤖

iCubLisboa01

### Request/Failure description

Right thumb Tactile sensor is not responding to any touch in the skin gui

### Detailed context

The right thumb tactile sensor doesn't respond to any stimuli(green bubble), we have tried to change the calibration settings and checking all of the wires that run from the sensor to the FTC_00 board(cyan line), the chips on the board itself have also been swapped around to check if those where causing any issue(magenta square).

[image]

All other sensors in the hand seem to be work well. And we can read the raw data output but it's not very human readable since all the sensors of the hand are shown at once.

What would be some recommend ways to diagnose what is the particular problem and how to maybe solve them.
My own suggestion is to try swap 2 finger tips and check if the thumb works with a known working one and see if the thumb now responds.

As a side question to this issue, what does the FTC_00 board do and what to the smaller chips mounted to it do?

AntonioConsilvio, 2023-01-31:
Hi @Ryo-Naohara, the fingertip works like this: there is the fingertip sensor (green in your picture), then the (cyan) FTC_connector (magenta) cable that connects the fingertip to the FTC board. 

So the FTC connector (magenta) is just a bridge to connect the fingertip to the FTC board. The FTC connector is useful because it makes it easier to detach the fingertip from the FTC board, which connects all the fingertip sensors for reading values.

Therefore, if you are sure that the cable is in place, you can perform these operations:

- You need to make sure that the FTC (magenta) connector is well connected to the FTC board.

- As suggested, you can try desoldering the thumb fingertip and swapping it with the index fingertip. (You can only swap the thumb with the index finger, because the fingertip cards all look the same, but they have an address to read the value, and the thumb and index finger have the same address, so you can read the value even if you change the fingertip. When you desolder the fingertips, you need to disconnect the FTC connector from the FTC board).

- I do not understand if you have already done this, but you can try changing the position of the FTC connector on the FTC board (always thumb with forefinger).

If, by swapping the thumb fingertip with the index finger, the thumb still does not work, it means that the fingertip sensor is not working.

If instead, the index finger does not work, it means the problem is in the cable or FTC connector.

If you have any question, write to us.

cc @Fabrizio69

Ryo-Naohara, 2023-01-31:
Hello, i am quite confused by this.

"If, by swapping the thumb fingertip with the index finger, the thumb still does not work, it means that the fingertip sensor is not working.

If instead, the index finger does not work, it means the problem is in the cable or FTC connector."

Wouldn't it be the opposite where if by swapping the fingertips the thumb doesn't work it means that the cable or connector is broken. 
And if the thumb does work with the new finger tip it means that the finger tip sensor is broken?

AntonioConsilvio, 2023-02-02:
Hi @Ryo-Naohara, It should be as you said, if the thumb was the same as the index finger. 

But, in theory, the Thumb Fingertip, even if it is moved to the Index finger, the sensor remains the Thumb Fingertip Sensor. 

So, when you swap the fingertips, the Index Fingertip Sensor remains the Index Fingertip Sensor even though it is placed on the thumb, and the Thumb Fingertip Sensor remains the Thumb Fingertip Sensor, even though it is placed on the index finger.

It is possible to swap fingertips because they have the same address for data transmission, but they are mapped to ensure a better reading of the values, so the thumb should remain thumb and the same is true for the index finger, even if they change position. 

So if I say Thumb Fingertip Sensor, I am referring to a specific fingertip, regardless of its current location.

Anyway, Have you tried test? How did it go?

If you have any question, write to us.

cc @sgiraz

Ryo-Naohara, 2023-02-02:
Hello, if i am reading this correctly this means that if i swap them and test them, when i touch the thumb, the index should be the one reacting and when i touch the index it should be the thumb that's reacting?

If this is true I am very confused from the results since when i swapped them the thumb fingertip that is now in the index finger didn't react in the program but the index fingertip in the thumb finger did react has the thumb in the program.

Ryo-Naohara, 2023-02-02:
[image]

AntonioConsilvio, 2023-02-02:
> Hello, if i am reading this correctly this means that if i swap them and test them, when i touch the thumb, the index should be the one reacting and when i touch the index it should be the thumb [...]

@Ryo-Naohara I'm not talking about the name you can see on the fingertip testing program. I'm talking about the true fingertip position.

So if you move the Thumb Fingertip Sensor to the index finger, the program will call it "Index," but the true fingertip position is always the thumb.

So the Thumb Fingertip Sensor, even if it placed on the index finger, is still not working, is that correct?

In this case it is the Thumb Fingertip Sensor (which is currently located on the index finger, so in the program it will be called "index") that is no longer working, so a new one is needed, is that correct? 

If I have not been clear enough write to me

cc @sgiraz

AntonioConsilvio, 2023-02-02:
> [image]

@Ryo-Naohara Ok, so the problem is the Sensor, and you need a new one!

cc @sgiraz

Ryo-Naohara, 2023-02-02:
Since the left hand is being sent back for repairs, could i possibly replace the left thumb with the right one?

AntonioConsilvio, 2023-02-02:
@Ryo-Naohara I am sorry, but you can't, because the solder pads in the left-hand Fingertips are located on the opposite side from the right-hand ones, and so the wires don't reach them.

Ryo-Naohara, 2023-02-02:
So we would need to order a new thumb fingertip sensor?

AntonioConsilvio, 2023-02-03:
Hi @Ryo-Naohara, It is not necessary to order a new fingertip.

If you agree, we will send you a new fingertip when we ship the left forearm back to you. So that we can make only one shipment.

cc @sgiraz

plinioMoreno, 2023-02-03:
Hi @AntonioConsilvio ,

We agree with your plan. We are going to get things ready here to send the arm to you soon.

sgiraz, 2023-05-12:
Maintenance activity completed, and the new thumb fingertip has been shipped, see:
- https://github.com/robotology/icub-tech-support/issues/1474#issuecomment-1543936332

Closing.

sgiraz, 2023-06-06:
Hi @plinioMoreno, is there any reason for re-opening this issue?

plinioMoreno, 2023-06-06:
Hi @sgiraz, I reopened by mistake. In fact @Ryo-Naohara replaced the cable that connect the tactile sensor of the right finger tip and it is working properly now.

closed 2023-06-06 by plinioMoreno