robotology/mesh-support #1496: Right wrist tendon out of place

S-Dafarra, 2023-02-02:
### Device name 🤖

iCubGenova09

### Request/Failure description

One of the wrist tendons is out of place, probably damaged.

### Detailed context

During the visit of the Osaka group, there were a lot of energic handshakes and one tendon broke.

[image]

sgiraz, 2023-02-02:
Hi @S-Dafarra,

@GiacomoVassallo13 is going to take a look at it today.

AntonioConsilvio, 2023-02-02:
Hi @S-Dafarra, @GiacomoVassallo13 replaced the steel cable.

When you can, please send us feedback on the operation of the robot.

cc @sgiraz

S-Dafarra, 2023-02-03:
Hi @AntonioConsilvio, it seems functional! Thanks

closed 2023-02-03 by AntonioConsilvio