robotology/mesh-support #1497: Broken tendon in right thumb (does not calibrate distal thumb)

GiuliaScorzaAzzara, 2023-02-07:
### Device name 🤖

iCubGenova07

### Request/Failure description

During the initial calibration, the robot does not calibrate the distal thumb in the right hand. 

### Detailed context

We noticed that the tendon is broken (i.e., the cable is cut). It is necessary to replace the cable.

sgiraz, 2023-02-07:
Hi @GiuliaScorzaAzzara,

We gonna check it asap.
Stay tuned!

sgiraz, 2023-02-10:
The maintenance activity is planned for Wednesday, 15 at 2 p.m.

AntonioConsilvio, 2023-02-15:
Hi @GiuliaScorzaAzzara, @GiacomoVassallo13 and I replaced the distal tendon of the thumb. We tested the robot and it seems to work without any problems.

https://user-images.githubusercontent.com/114915464/219090033-df4ee2e8-88c1-4d86-b097-761ba0d2a938.mp4

cc @sgiraz

closed 2023-02-15 by AntonioConsilvio