robotology/mesh-support #1498: Face expression stopped to work

lrapetti, 2023-02-09:
### Device name 🤖

ergoCub

### Request/Failure description

Face expression of the robot stopped to work.

### Detailed context

In order to enable wbd on `ergocub-torso`, we have installed the robotology-superbuild dependencies using the [`install_apt_dependencies.sh`](https://github.com/robotology/robotology-superbuild/blob/master/scripts/install_apt_dependencies.sh) script. While doing this, however, the following message was raised:
[image]
we continued, but after a while we realized the face expressions stopped to work (I guess it was the service that was supposed to be restarted). After restarting the robot motors and CPU, the face expression seems to be still off:
[image]

cc @GiulioRomualdi @mebbaid

pattacini, 2023-02-09:
@AntonioConsilvio reported that by unplugging and then plugging the HMDI back again solved the problem.
We should better inquire about the root cause of this.

lrapetti, 2023-02-10:
Thanks for checking, for me the issue can be close and we will monitor if it happens again!

closed 2023-02-10 by lrapetti