robotology/mesh-support #1500: iCubGenova09 right wrist tendon not taut/broken

S-Dafarra, 2023-02-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

One of the right wrist tendons appears to be either broken or not taut

### Detailed context

[image]

### How does it affect you?

Tomorrow morning (Feb 10) we have a demo and we need the robot for the teleoperation

cc @DanielePucci @Fabrizio69 @maggia80 @GiacomoVassallo13 @GiulioRomualdi @lrapetti

AntonioConsilvio, 2023-02-10:
Hi @S-Dafarra, the problem has been temporarily resolved, however, this wrist problem will recur because this joint needs specific intervention to be fixed permanently.

Therefore, we should plan if and when to do it.

cc @S-Dafarra

S-Dafarra, 2023-02-10:
For the demo sake, the fix was good enough. At the moment, we have no other demos planned and soon enough the robot hands will be unmounted. So, I am not sure if it is worth doing that intervention now.

S-Dafarra, 2023-02-16:
> Hi @S-Dafarra, if you confirm that you won't have any more demos with iCub3 before disassembling the hands, we don't need to do the intervention and we can close this issue.
> 
> cc @sgiraz

Hi @AntonioConsilvio, I am not aware of any forthcoming demos with iCub3. Clearly, things can change at any moment, but for the time being I think we can close this issue.

closed 2023-02-16 by AntonioConsilvio