robotology/mesh-support #1501: Right shoulder loses calibration over time

GiulioRomualdi, 2023-02-14:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While testing we noticed that the right shoulder loses calibration from time to time.

### Detailed context

As far as I know, @AntonioConsilvio already knows the issue and it seems to be related to the fact that the incremental calibration is missing

sgiraz, 2023-02-15:
[image]

DanielePucci, 2023-02-16:
@sgiraz 
> Tomorrow @davidetome and I (maybe with the help of @ale-git) will try to apply the incremental calibration as done in `iCub3`.

Is this still confirmed for today? The loss of the arm calibration is making complex the preliminary experimental activities since we need to reboot the robot and change the configuration files manually from times to times during the day

pattacini, 2023-02-16:
Hi @DanielePucci 

@sgiraz @davidetome @ale-git and @AntonioConsilvio are done with the incremental calibration of the right arm. They're now dealing with the left arm.

pattacini, 2023-02-16:
Incremental calibration completed!
Thanks to @ale-git @sgiraz @davidetome 👍 

New values in conf files updated via:
- https://github.com/robotology/robots-configuration/pull/440

closed 2023-02-16 by pattacini